RavenSystem/esp-homekit-devices #1337: WiFi SSID truncate and impossible to connect during auto update

aleroot, 2021-02-13:
Hi,
I have a WiFi network from the provider router with name: `533C Hyperoptic 1Gb Fibre 2.4Ghz`, when it is saved by HAA setup tool the last character is truncated and so it is impossible to connect to the WiFi ...

Log :

```
no 533C Hyperoptic 1Gb Fibre 2.4Gh found, reconnect after 1s
reconnect
scandone
```

aleroot, 2021-02-13:
Looks like the issue is in the sdk_bss_info definition in: https://github.com/SuperHouse/esp-open-rtos/blob/503e66a500419e8863998b7ea784c5e26a7a5f7c/include/espressif/esp_sta.h it is defined as 32 char array while it should be 33 since it is null terminated.

RavenSystem, 2021-02-13:
Forget PR. I will fix it by myself:
```
Jan 01 00:00:03 connected with 533C Hyperoptic 1Gb Fibre 2.4Ghz, channel 9
Jan 01 00:00:03 dhcp client start...
```

closed 2021-03-17 by stale[bot]