RavenSystem/esp-homekit-devices #1339: 4.9.0 Shelly 1 connection to my wifi

adamkasprzak, 2021-02-16:
Hi! 

I have flashed over 30 devices in the past and never had any problems. 

Today I have tried to flash another Shelly 1 with the newest version (4.9.0) and older one (4.3.0). I have a problem with connecting to my wifi (Legacy Support checked, 2.4 + 5GHz). Once I put my wifi name or select from the list + password (for sure correct) it behaves as there is no wifi settings or password is incorrect.

Do I miss something?

adamkasprzak, 2021-02-16:
Just checked Shelly 2.5 with 4.9.0 - the same problem.

adamkasprzak, 2021-02-16:
Another check - old Sonoff basic with saved settings connected without any problems to the network. It seems that either these two units are somehow broken. I'm using the latest version 4.9.0 and flashing with fullhaaboot.bin using esptool.

mariusvasss, 2021-02-16:
Flash haaboot.bin, not fullhaaboot.bin

adamkasprzak, 2021-02-16:
Some logs:

Sysparam OK

Wifi Init
mode : sta(8c:aa:b5:47:e3:20)
add if0
Saved BSSID: fcecdab70d97
Wifi Mode: Normal

HAA OTA - SETUP MODE

mode : sta(8c:aa:b5:47:e3:20) + softAP(8e:aa:b5:47:e3:20)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-47E320
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
add 0
aid 25
cnt

connected with <MYWIFINAME>, channel 11
dhcp client start...
ip:192.168.XX.XX,mask:255.255.255.0,gw:192.168.XX.XX

It seems it is connected to my wifi and creating a setup wifi?

adamkasprzak, 2021-02-16:
> Flash haaboot.bin, not fullhaaboot.bin

Acc to this https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation I should flash fullhaaboot...

adamkasprzak, 2021-02-16:
OK, so I figured that connection to my wifi is established, but Shelly creates also own network at the same time - the unit is constantly, all the time in setup mode.

RavenSystem, 2021-02-17:
GitHub server side issue:
https://github.com/HomeACcessoryKid/life-cycle-manager/issues/23#issuecomment-780155826

adamkasprzak, 2021-02-17:
Thank you for explanation. Could you please explain why older, configured devices work and this new ones are stuck?

jakudo, 2021-02-17:
@adamkasprzak works now :)

closed 2021-02-17 by adamkasprzak