RavenSystem/esp-homekit-devices #1340: Loop by installing HAA

jRaufe, 2021-02-16:
Hi,
Trying to install HAA in a Shelly 1, I’m in a loop!

_> iPad Pro de Jose $ nc -kulnw0 45678 
> 
> Advanced ESP Logger (c) 2020 José Antonio Jiménez Campos
> 
> 0003.732 192.168.1.126-A9A494: DNS check result  = OK
> 0009.080 192.168.1.126-A9A494: DNS..OK Socket..OK Connect..OK SSL..OK set_fd to github.com port 443..OK
> 0009.085 192.168.1.126-A9A494: sent OK
> ossorigin=anonymous; as=script,<https://github.githubassets.com/assets/behaviors-bf28db4d.js>; integrity=sha512-vyjbTdEGTiemoiT41KyVSh3+NxJv285wa1qlccyGrTsQsAAvhBSUrT0t1LZbJVEfccSTx+zLAyTekiljOX2wKA==; rel=preload; crossorigin=anonymous; as=script,<https://github.githubassets.com/assets/environment-f0adafbf.js>; integrity=sha512-8K2vvwbW+6H27Nad5ydg8PA2/aMD/LKq+EiK9s0U0hhVZxCI2tWBsYk9beAtisRw2j+Or5k2/F+6dk02nmj/PA==; rel=preload; crossorigin=anonymous; as=script
> 0011.309 192.168.1.126-A9A494: Vary: X-PJAX, Accept
> 
> 0011.311 192.168.1.126-A9A494: 
> 0011.312 192.168.1.126-A9A494: HTTP returns 302
> 
> 0011.312 192.168.1.126-A9A494: 
> 0011.316 192.168.1.126-A9A494: !!! SERVER ERROR
> 
> 0011.317 192.168.1.126-A9A494: 
> 0011.317 192.168.1.126-A9A494: !!! Error downloading OTAMAIN signature
> 
> 0011.318 192.168.1.126-A9A494: 
> 
> 0016.328 192.168.1.126-A9A494: 
> 0016.329 192.168.1.126-A9A494: *** STARTING UPDATE PROCESS
> 
> 0016.329 192.168.1.126-A9A494: 
> 
> 0016.329 192.168.1.126-A9A494: 
> 0016.330 192.168.1.126-A9A494: Running HAABOOT
> 
> 0016.330 192.168.1.126-A9A494: 
> 0016.330 192.168.1.126-A9A494: HomeKit data migration...
> 0016.331 192.168.1.126-A9A494: Data is already migrated
> 
> 0016.332 192.168.1.126-A9A494: 
> 0016.332 192.168.1.126-A9A494: Get sign
> 
> 0016.332 192.168.1.126-A9A494: 
> 0016.333 192.168.1.126-A9A494: DOWNLOADING FILE
> 
> 0016.333 192.168.1.126-A9A494: 
> 0016.334 192.168.1.126-A9A494: Forwarding: github.com/RavenSystem/haa/releases/latest/download/otamain.bin.sec
> 0016.337 192.168.1.126-A9A494: -----
> 0016.338 192.168.1.126-A9A494: GET /RavenSystem/haa/releases/latest/download/otamain.bin.sec HTTP/1.1
> 0016.344 192.168.1.126-A9A494: Host: github.com
> 0016.346 192.168.1.126-A9A494: Range: bytes=0-1
> 
> 0016.347 192.168.1.126-A9A494: 
> 
> 0016.328 192.168.1.126-A9A494: 
> 0016.329 192.168.1.126-A9A494: *** STARTING UPDATE PROCESS
> 
> 0016.329 192.168.1.126-A9A494: 
> 
> 0016.329 192.168.1.126-A9A494: 
> 0016.330 192.168.1.126-A9A494: Running HAABOOT
> 
> 0016.330 192.168.1.126-A9A494: 
> 0016.330 192.168.1.126-A9A494: HomeKit data migration...
> 0016.331 192.168.1.126-A9A494: Data is already migrated
> 
> 0016.332 192.168.1.126-A9A494: 
> 0016.332 192.168.1.126-A9A494: Get sign
> 
> 0016.332 192.168.1.126-A9A494: 
> 0016.333 192.168.1.126-A9A494: DOWNLOADING FILE
> 
> 0016.333 192.168.1.126-A9A494: 
[... 235 lines cut ...]
> 
> 0012.070 192.168.1.126-A9A494: 
> 0012.070 192.168.1.126-A9A494: -----
> 
> 0012.072 192.168.1.126-A9A494: 
> 0012.073 192.168.1.126-A90013.598 192.168.1.126-A9A494: DNS..OK Socket..OK Connect..OK SSL..OK set_fd to github.com port 443..OK
> in=anonymous; as=script,<https://github.githubassets.com/assets/behaviors-bf28db4d.js>; integrity=sha512-vyjbTdEGTiemoiT41KyVSh3+NxJv285wa1qlccyGrTsQsAAvhBSUrT0t1LZbJVEfccSTx+zLAyTekiljOX2wKA==; rel=preload; crossorigin=anonymous; as=script,<https://github.githubassets.com/assets/environment-f0adafbf.js>; integrity=sha512-8K2vvwbW+6H27Nad5ydg8PA2/aMD/LKq+EiK9s0U0hhVZxCI2tWBsYk9beAtisRw2j+Or5k2/F+6dk02nmj/PA==; rel=preload; crossorigin=anonymous; as=script
> 0013.781 192.168.1.126-A9A494: Vary: X-PJAX, Accept
> 
> 0013.782 192.168.1.126-A9A494: 
> 0013.784 192.168.1.126-A9A494: HTTP returns 302
> 
> 0013.785 192.168.1.126-A9A494: 
> 0013.790 192.168.1.126-A9A494: !!! SERVER ERROR
> 
> 0013.792 192.168.1.126-A9A494: 
> 0013.792 192.168.1.126-A9A494: !!! Error downloading OTAMAIN signature
> 
> 0013.793 192.168.1.126-A9A494: 
> 
> 0013.821 192.168.1.126-A9A494: 
> 0013.823 192.168.1.126-A9A494: Restarting...
> _

Any idea what to do?

RavenSystem, 2021-02-17:
GitHub server side issue:
https://github.com/HomeACcessoryKid/life-cycle-manager/issues/23#issuecomment-780155826

nickforev, 2021-02-17:
Work it now!

jRaufe, 2021-02-18:
👍

closed 2021-02-18 by jRaufe

weishiji, 2021-09-28:
> Work it now!

I met the same problem, how to solve it?