RavenSystem/esp-homekit-devices #1341: Sonoff S26 - can't add to HomeKit

jakudo, 2021-02-17:
Hi,

I have a problem with Sonoff S26 - Smart Plug. I flashed it as usually, connected to my WiFi and enter JSON (tried several JSONs). It connects to the WiFi (can ping it), however, I cannot add it to HomeKit. I've waited for about 15 minutes and still nothing. Also there is a strange thing - usually when I flash any sonoff device and connect it to my WiFi, it stops broadcast the HAA-xxxxxx wifi. Now it still broadcasting this wifi I can connect to it and enter the setup mode.

Any idea?

tonysprenk, 2021-02-17:
https://github.com/HomeACcessoryKid/life-cycle-manager/issues/23#issuecomment-780155826

jakudo, 2021-02-17:
Thanks, not sure what that means, however, I understand that there is some problem and it is in progress. Thanks.

adamkasprzak, 2021-02-17:
I have the same problem. We need to wait for the fix. Btw, @RavenSystem  - thank you for quick responses and magnificent work!

nickforev, 2021-02-17:
Work it now!!!

closed 2021-02-17 by jakudo