RavenSystem/esp-homekit-devices #1342: Home vpn

brkysnol, 2021-02-17:
I set up a vpn service to access the home network from the outside. How do I manage connected apple home kit devices at home with vpn ?

tonysprenk, 2021-02-17:
You dont have a home hub (Apple TV, ipad, HomePod)?

brkysnol, 2021-02-17:
I'm afraid I don't have any of this. I only have a vpn on my home network.

brkysnol, 2021-02-17:
> You dont have a home hub (Apple TV, ipad, HomePod)?

I'm afraid I don't have any of this. I only have a vpn on my home network.

adamkasprzak, 2021-02-17:
I ended up using home hub. Firstly iPad, then ATV. Homekit doesn't work via VPN in my case.

brkysnol, 2021-02-17:
> Ana hub'ı kullanmayı bıraktım. Önce iPad, ardından ATV. Homekit benim durumumda VPN üzerinden çalışmıyor.

I don't have any of this :( how to access it remotely.

jakudo, 2021-02-17:
I believe that the VPN doesn't help with your problem. You need some hub (Apple TV, iPad, HomePod). Without it you cannot use homekit remotely.

tonysprenk, 2021-02-17:
It can be done but it’s not easy. HomeKit uses Bonjour which does notebook over vpn. You need to setup some service to relay Bonjour messages. Easier would be to buy a cheap Apple TV or iPad.

peros550, 2021-02-17:
Technically you could enable insecure mode in HAA ("u":1) and find a way to control accessories using http get requests from your mobile. This could work over vpn.

I'm not endorsing insecure mode, however it's there if you want to use it. 

But, with an Apple TV you can do so much more

brkysnol, 2021-02-18:
> Yapılabilir ama kolay değil. HomeKit, dizüstü bilgisayarı vpn üzerinden yapan Bonjour kullanır. Bonjour mesajlarını iletmek için bazı servisler kurmanız gerekir. Daha kolay, ucuz bir Apple TV veya iPad satın almak olacaktır.

> Teknik olarak HAA'da güvenli olmayan modu etkinleştirebilir ("u": 1) ve cep telefonunuzdan http alma isteklerini kullanarak aksesuarları kontrol etmenin bir yolunu bulabilirsiniz. Bu vpn üzerinden çalışabilir.
> 
> Güvensiz modu onaylamıyorum, ancak kullanmak istiyorsanız orada.
> 
> Ancak bir Apple TV ile çok daha fazlasını yapabilirsiniz

Thank you for your answers, I want to do what you say. Are there any sources ?

closed 2021-03-21 by stale[bot]