RavenSystem/esp-homekit-devices #1345: Window Covering: Status inputs do not work

jubudu, 2021-02-18:
I want to set the status of my windows coverings using GPIO4 without acting the relay-actions, because the status of the accessory might not be the real one in some scenarios. A long story...
 
The configuration I use is shown below:

` "a": [{
    "t": 45, "o": 10, "c": 7, "f": 70, "cd": 2,
    "0": { "r": [{ "g": 15, "v": 0 }, { "g": 12, "v": 1 }] },
    "1": { "r": [{ "g": 15, "v": 1 }, { "g": 12, "v": 0 }] },
    "2":  { "r": [{ "g": 15, "v": 0 }, { "g": 12, "v": 0 }] }
    "g0": [{ "g": 4,  "t": 0 }],
    "g1": [{ "g": 4,  "t": 1 }]
  }]`

But the status inputs g0 and g1 are not processed at all, neither physically nor in the output-log.

Thanks for help or work around...

RavenSystem, 2021-02-26:
`"g0"`and `"g1"` are not processed because window cover doesn't support it:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering

closed 2021-02-26 by RavenSystem

jubudu, 2021-03-02:
Thank you, but I am confused, now. Cited from the wiki-link referring to window covering:

"State and Status Inputs

State inputs "f[n]" & Status Inputs "g[n]" are supported by this accessory."

RavenSystem, 2021-03-02:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering#State-and-Status-Inputs

Supported list includes from f0 to f6, nothing more.

jubudu, 2021-03-10:
I refer to the uppermost table... Anyway, it would be a nice feature ;-)

[image]