RavenSystem/esp-homekit-devices #1346: Dimmer with inverse PWM?

pnuding, 2021-02-19:
I have a single white channel PWM dimmable controller here, where the PWM output is inverted, so it attenuates the light. 
I am not sure how to map this correctly - I tried with a "fa":[-1] to see if that inverts the color, but that doesn't do anything. 
What would be the right way?

amanbaig123, 2021-02-21:
Adding a value of 100 to GPIO number will declare it as inverted.
Check https://github.com/RavenSystem/esp-homekit-devices/wiki/Lightbulb#rgbw-gpio-channels

pnuding, 2021-02-21:
Oh my - I should learn to read! 😆 
Thank you so much, works like a charm with this!

closed 2021-02-21 by pnuding

planevina, 2022-06-15:
> Adding a value of 100 to GPIO number will declare it as inverted. Check https://github.com/RavenSystem/esp-homekit-devices/wiki/Lightbulb#rgbw-gpio-channels
==================
Forgive my poor English
I have the same problem
But I can not understand "Adding a value of 100 to GPIO number will declare it as inverted."

for example ,I have a single channel pwm lightbulb
the config json is       "g":[3]   (3 is gpio num.)

"add a value to gpio number " means    
   "g":[ 3 : 100 ]    
or
  "g":[ "3" : 100 ]    

?

thanks