RavenSystem/esp-homekit-devices #1348: LDR light sensor code correction

peros550, 2021-02-21:
hi @RavenSystem I read your code, and also your reference [here](https://www.allaboutcircuits.com/projects/design-a-luxmeter-using-a-light-dependent-resistor/) and it looks the calculation of LDR's resistance isn't in line to the above article.  I went through all the calculations and using the following circuit: 

[image] 

the ldr_resistor should be calculated like that: 

```
 int raw_measurment = sdk_system_adc_read();
 ldr_resistor = ((1023.1 - (float)raw_measurment)/(float)raw_measurment) * LIGHT_SENSOR_RESISTOR ;
```

This is my json: 

{
  "c": {"t":14,"u":1,"o":4
  },
  "a": [{"t":50,"n":0,"f":14357564.33,"po":1.396066767,"re":8550},{"1":{"s":[{"a":1}]},"t":1,"b":[],"s":0}]
}

The above two (json + new code) give me pretty good results. 

Because I guess, you would like to provide a lux calculations for circuits with positioned LDR and R inverted, I did the calculations and your code line 3260 should be something like this: 

```
  int raw_measurment = sdk_system_adc_read();
        if (LIGHT_SENSOR_TYPE == 0) {
            //ldr_resistor = LIGHT_SENSOR_RESISTOR * ldr_voltage;   <-- Wrong
           ldr_resistor = ((1023.1 - (float)raw_measurment)/(float)raw_measurment) * LIGHT_SENSOR_RESISTOR ;     //new

        } else {
           // ldr_resistor = LIGHT_SENSOR_RESISTOR / ldr_voltage; <-- Wrong
            ldr_resistor = (LIGHT_SENSOR_RESISTOR *  (float)raw_measurment) / (1023.1 - (float)raw_measurment)
        }
```

RavenSystem, 2021-02-21:
Great! This was something I had to do. Many thanks!

RavenSystem, 2021-02-21:
https://github.com/RavenSystem/esp-homekit-devices/releases/tag/HAA_5.0.4

closed 2021-02-21 by RavenSystem