RavenSystem/esp-homekit-devices #1349: V5.0.4: "No Response" and DHT22 error

jubudu, 2021-02-22:
Dear all, with the newest release 5.0.4 I observe:
- For most of the time Home App is showing "no response" for the windows covering and temperature sensor accessory after **successful** pairing. 
- If accessory is answering some times, the temp and DHT-Values are always "0" and after any action it returns to "no response.
- **The same hardware (ElectroDragon 2 Relais board) is working using V4.9.0.** 

Here is my config for V5.0.4:

{
"c":{"l":16,"i":0,"o":1,
    "n": "THauptrollos-880534",
    "b":[{"g":2,"p":0,"t":5}],
    "0":{"r":[{"g":12,"g":13}] },
    "2": {  "r": [{ "g": 16, "v": 1, "i": 3 }] },
    "3": {  "r": [{ "g": 16, "v": 1, "i": 10 }] },
    "4": {  "r": [{ "g": 16, "v": 1 }] },
    "5": {  "r": [{ "g": 16, "v": 1 }] },
    "ntp": "ntp1.t-online.de",
    "tz" : "CET-1CEST-2,M3.5.0/2,M10.5.0/2"
     },
"a":[
     {
    "t":45,
    "o":10,
    "c":5,
    "f":0,
    "s": 5,
    "0":{"r":[{"g":12,"v":1,"i":1},{"g":13}]},
    "1":{"r":[{"g":12},{"g":13,"v":1,"i":1}] },
    "2":{"r":[{"g":12},{"g":13}]},
    "3":{"r":[{"g":12,"i":1},{"g":13}]},
    "4":{"r":[{"g":12},{"g":13,"i":1}]},
    "5":{"r":[{"g":12},{"g":13}], "m":[[2, 0]] },
    "6":{"r":[{"g":12},{"g":13}], "m":[[2, 100 ]] },
    "f0":[{"g":5,"p":0}],
    "f1":[{"g":4,"p":0}],
    "f2":[{"g":5,"t":0},{"g":4,"t":0}]
    },
    {
    "t":45,
    "o":10,
    "c":5,
    "f":0,
    "s": 5,
    "0":{"r":[{"g":12,"v":1,"i":0.3},{"g":13}]},
    "1":{"r":[{"g":12},{"g":13,"v":1,"i":0.3}]},
    "2":{"r":[{"g":12},{"g":13}]},
    "3":{"r":[{"g":12,"i":0.3},{"g":13}]},
    "4":{"r":[{"g":12},{"g":13,"i":0.3}]},
    "5":{"r":[{"g":12},{"g":13}]},
    "6":{"r":[{"g":12},{"g":13}]},
    "f0":[{"g":5,"p":0}],
    "f1":[{"g":4,"p":0}],
    "f2":[{"g":5,"t":0},{"g":4,"t":0}]
    },
    {
    "t": 24,
    "g": 14,
    "n": 2,
    "j": 30,
    "z": -3.4,
    "h": 5.0
    }
    ]
}

Thanks for your comments and help! Jubudu

ymyshko, 2021-02-24:
I got the same result with DHT22. I reset the config and applied the saved one. Everything works.

jubudu, 2021-02-24:
> I got the same result with DHT22. I reset the config and applied the saved one. Everything works.

Thanks for your reply. 
Did you test with HAA V5.04?
Can you please post your saved config, so I can give it a try?

jubudu, 2021-02-27:
Today I have investigated on the "No Response" problem using V5.08:
It seems to depend from the windows covering accessory:
If two are configured, as in the JSON above, NTP fails, and I get lots of "! [n] Reading from socket (113). Closing" errors.
If only one is configured, everything works fine.

jubudu, 2021-02-28:
Same test with V5.0.10:
"No Response" an no NPT, when two windows covering are configured. Reading from socket errors in the log-file:

Jan 01 01:00:15 >>> mDNS_announce: working with TTL 4500
Jan 01 01:00:15 >>> mDNS_announce: sent
Jan 01 01:00:19 NTP update (-1)
Jan 01 01:00:27 NTP update (-1)
Jan 01 01:00:35 NTP update (-1)
...
Jan 01 01:01:51 ! [6] Reading from socket (113). Closing
Jan 01 01:01:51 [5] Get Acc
Jan 01 01:01:52 [6] Closed
Jan 01 01:01:52 [10] Verify 1/2
Jan 01 01:01:53 ! [9] Reading from socket (104). Closing
Jan 01 01:01:53 ! [7] Reading from socket (104). Closing
Jan 01 01:01:53 ! [5] Reading from socket (104). Closing
Jan 01 01:01:53 [9] Closed

APIL0T, 2021-03-03:
Same issue,I have 10 sonoff mini configured, and the same problem! several times a day the all  get "no response"

closed 2021-04-05 by stale[bot]