RavenSystem/esp-homekit-devices #1350: Sonoff Dual Lock Json

asbesto01, 2021-02-24:
Hello everybody,

can someone create for me a json to open a lock for 0.6 seconds, then relock it?

Thank you so much!!

tonysprenk, 2021-02-24:
This example is from the [wiki](https://github.com/RavenSystem/esp-homekit-devices/wiki/Lock-Mechanism#example-of-single-lock-mechanism) but I changed “i” to 6: 
```

{
  "a":[{
    "t":4,
    "i":3,
    "0":{"r":[{"g":4,"v":1,"i":6}]},
    "1":{"r":[{"g":4}]},
    "b":[{"g":5,"p":0,"i":1,"t":0}]
  }]
}

```

the lock mechanism is connected to gpio 4 and it has a button connected to gpio 5.

asbesto01, 2021-02-24:
{
   "c":{
      "o":1,
      "l":13,
      "i":1
   },
   "a":[
      {
         "t":4,
         "i":0.4,
         "0":{
            "r":[
               {
                  "g":12,
                  "v":1
               }
            ]
         },
         "1":{
            "r":[
               {
                  "g":12,
                  "v":0
               }
            ]
         },
                  "1":{
            "r":[
               {
                  "g":5,
                  "v":1
               }
            ]
         },
         "b":[
            {
               "g":9
            }
         ]
      }
   ]
}

this is the right json, I wrote it by myself and it works, I am so proud!!!!!

closed 2021-03-29 by stale[bot]