RavenSystem/esp-homekit-devices #1352: Can not achieve maximal power on both warm and cold lightbulb channels on Shelly RGBW2 with HAA 5.0.6.

vsevolod-volkov, 2021-02-26:
Hi!

I want to control two independent while led stripes with Shelly RGBW2 using HAA 5.0.6 lightbulb accessory as shown below:
`{
   "a": [
      {
         "t":30,
         "n": 2,
         "g": [14, 4],
         "b": [ {"g":5, "p":0}, {"g":13} ]
      }
   ]
}`

Both LED stripes actually have the same colour temperature and grouped together in warm/cold pair for ease of control.
Either "cold" or "warm" line can glow with full power, but I can not get both LED stripes glow with full power simultaneously mixing both "colours" with HomeKit.
I achieved that in pre-5.0 HAA versions using "fv": 2.0, "fw": 2.0 factors, but can not find their analog in 5.0.x.
Could you please explain how do I achieve that in HAA 5.0.6. Thanks.

RavenSystem, 2021-02-26:
Use flux array: `"fa":[2,2]`

https://github.com/RavenSystem/esp-homekit-devices/wiki/Lightbulb#Flux-array

vsevolod-volkov, 2021-02-28:
It works. Thanks.

closed 2021-02-28 by vsevolod-volkov