RavenSystem/esp-homekit-devices #1353: Doorbell json required

okanduzyel, 2021-02-27:
Hi,

here https://github.com/RavenSystem/esp-homekit-devices/issues/1144 you discussion about a doorbell json example. I tried that but After the installation of HAA, the device doesn't respond. 

Can you share a doorbell example json?

okanduzyel, 2021-03-01:
`{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":13,"f0":[{"g":14,"t":0}]}]}`

This is your example that it is okay. But It is unfortunately just a stateless button.

closed 2021-03-01 by okanduzyel

mariusvasss, 2021-03-04:
> 
> 
> Hi,
> 
> here #1144 your discussion about a doorbell json example. I tried that but After the downloading of HAA, the device doesn't respond. I think, the json has problem. Because I tried other your json examples. They are okay.
> 
> Can you share a json of doorbell example?

Sonoff mini doorbell with buton on gpio 4, chime to gpio 12, upgrade classic doorbell 

{"c":{"n":"Doorbell","l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},"a":[{"t":13,"f0":[{"g":4,"p":0,"i":1,"t":0}],"0":{"r":[{"g":12,"v":1,"i":0.5}]}}]}

Przylepzwm, 2021-03-04:
t: 13 is a doorbell, but I see it as a contact sensor. Is it correct?