RavenSystem/esp-homekit-devices #1354: Window-Covering

okanduzyel, 2021-02-28:
Hi, 

I have a window covering that has a step motor inside. I looked at your examples about window covering and found this https://github.com/RavenSystem/esp-homekit-devices/wiki/Window-Covering.

The motor driver of my window covering is a4988. It requires PWM input for each step changing of step motor. If I can add a PWM output that gives only output under opening-closing times, I can use your HAA.

How can I add it for that?

```
{
  "c": { "l": 0, "b": [{ "g": 2, "p": 0, "t": 5 }], "0": { "r": [{ "g": 15 }, { "g": 16 }] } },
  "a": [{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "cd": 2,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }] },
    "1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }] },
    "2": { "r": [{ "g": 15 }, { "g": 4 }] },
    "3": { "r": [{ "g": 15, "i": 0.2 }, { "g": 4 }] },
    "4": { "r": [{ "g": 15 }, { "g": 4, "i": 0.2 }] },
    "f0": [{ "g": 5, "i": 1, "p": 0, "t":0 }],
    "f1": [{ "g": 13, "i": 1, "p": 0, "t":0 }],
    "f2": [{ "g": 5 }, { "g": 13 }]
  }]
}
```

okanduzyel, 2021-02-28:
If I understand correctly, the window covering device type hasn't PWM output capability. 
Could you add it for me? Because there are a lot of solutions that use step motor...

pilot1981, 2021-02-28:
Are you trying to use HAA to control swing windows?

okanduzyel, 2021-02-28:
Nope. Roller blind curtain.

[image]

okanduzyel, 2021-03-02:
I tried to trigger via a notification to a light bulb which creates a single PWM. But this JSON, I put below, doesn't work. What am I missing?

```
{
  "c": { "q":1500, "l": 2, "b": [{ "g": 2, "p": 0, "t": 5 }], "0": { "r": [{ "g": 15 }, { "g": 16 }] } },
  "a": [{
    "t": 45,
    "o": 18,
    "c": 14,
    "f": 70,
    "cd": 2,
    "0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }], "m":[[ 2, -10001 ]]},
    "1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }], "m":[[ 2, -10001 ]]},
    "2": { "r": [{ "g": 15 }, { "g": 4 }], "m":[[ 2, -10000 ]]},
    "3": { "r": [{ "g": 15, "i": 0.2 }, { "g": 4 }], "m":[[ 2, -10001 ]]},
    "4": { "r": [{ "g": 15 }, { "g": 4, "i": 0.2 }], "m":[[ 2, -10001 ]]},
    "f0": [{ "g": 5, "i": 1, "p": 0, "t":0 }],
    "f1": [{ "g": 13, "i": 1, "p": 0, "t":0 }],
    "f2": [{ "g": 5 }, { "g": 13 }]
  },
  {
    "t":30,
    "it": [50],
    "g":[12],
    "b":[{"g":0}]
  }]
}
```

closed 2021-04-05 by stale[bot]