RavenSystem/esp-homekit-devices #1358: Troubleshooting “No Response”

andyallcorn, 2021-03-07:
Is there any advice for troubleshooting occasional “No Response” problems with HAA accessories? Eg. enabling logging and scouring it for particular telltale events?

I’m running 5.0.12 (though this is a long-standing problem) on a couple of Sonoff Minis and a Shelly 2.5, and have grabbed your HAA Manager app.

- Every couple of days one of these (could be any one) will go to No Response in HomeKit, while, according to router logs, remaining connected to WiFi the whole time.
- It’s an Asus AX82U, with a segregated 2.4-only SSID, with DHCP assigning a consistent IP. I’ve tried switching on or off various advanced features, none of which have made any discernible difference.
- I’ve tried settling TTL very low and this doesn’t seem to help.
- If I get the router to drop the device(s) off the network, they rejoin functioning perfectly.
- Other devices continue working fine, and after a few hours the No Response device will return to functioning too.

Bene2103, 2021-03-07:
I've experienced the same issue with different versions of HAA and different devices (as NodeMCU, MagicHome RGB and others)

The same as you did, I tried different settings in my router (FritzBox) with no success.

What I found is, when a device is not responding and I turn the device on via the button on the device, it becomes immediately responsive in HomeKit. Maybe this info is useful.

If we can provide more information or logs or anything to solve this issue let us know.

andyallcorn, 2021-03-08:
> What I found is, when a device is not responding and I turn the device on via the button on the device, it becomes immediately responsive in HomeKit. Maybe this info is useful.

That’s interesting - not observed that myself (sadly had the opportunity to try today when one of the switches moved to No Response), but I’ll keep a look out.

asbesto01, 2021-03-14:
I have a similar issue; my setup:

more than 30 sonoff devices on Eero mesh 3 nodes wifi.
I noticed that this happens to me only when I am on 3G or 4G outside home, while if I am at home with wifi enabled on iPhone, no issues

andyallcorn, 2021-03-23:
Had slightly different symptoms today, which is the first time I have had logging enabled. One accessory has been mostly No Response for the last 15 hours. It has occasional, fleeting periods of proper operation (around a minute), before returning to No Response.

During this time, two other accessories (without logging enabled) have remained responsive. All of them remain connected to wifi (according to router logs) the entire time.

Here's a recent section of log (no NTP server configured so excuse the timestamps). During this time the device (Shelley 2.5 running 5.0.14) has been almost entirely at No Response, but you will notice a couple of brief connections during which I managed to switch the light on and off.

Is there anything else I can provide to help with this?

[haa_output.txt](https://github.com/RavenSystem/esp-homekit-devices/files/6189753/haa_output.txt)

closed 2021-04-24 by stale[bot]