RavenSystem/esp-homekit-devices #1361: Simple pulse train to GPIO

lzparkma, 2021-03-10:
I have a device that needs to wake up and have a baud rate set by sending a pulse to a specific GPIO followed by a pulse train of three pulsed. Is active low so would be 4x loop of low-delay-high-delay.

Can multiple actions and delays be used?
Can I use UART and set a very low baud rate such that one bit is approx 250 ms and then send the hex equivalent of 0111101010?
Is the UART still limited to specifc GPIO via UART config?

Is there value for others in implementing simple binary pulse trains as an action type with three variables: idel state: high/low,  pulse duration: ms, pulse sequence in binary. Or adding delay into array or GPIO actions eg.

{
"c": {},
"a":[
    "1": { "r": [{ "g": 12, "v": 0,{"delay":250},{ "g": 12, "v": 1 },{"delay":250},{ "g": 12, "v": 1 }}] },
]
}

uluozgur, 2021-04-02:
Did you check the garage door example? https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door#garage-door-example

dbsqp, 2021-04-12:
Thanks, looks like it might be possible. Will give it a go, else will just roll separate trigger switch into my automation.

closed 2021-05-06 by RavenSystem