RavenSystem/esp-homekit-devices #1363: Automatically update?

okanduzyel, 2021-03-12:
Hi,

Does HAA automatically update itself when the new version released?

okanduzyel, 2021-03-13:
[image]

My question looks isn't clear, sorry. I flashed my esp device and adapted to my home and works now. I marked that checkbox before configured my device. Will it update itself when new version of HAA released?

pawelosin, 2021-03-21:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#auto-ota-updates

okanduzyel, 2021-03-21:
Ok. How can I learn my device's HAA version?

pawelosin, 2021-03-21:
It can be found in the device settings in the home app.

okanduzyel, 2021-03-21:
v5.0.14 released but my device still in v5.0.10. I marked Auto OTA update checkbox. Why didn't It update itself?

pawelosin, 2021-03-21:
Sorry, I don't know, but if it's a thermostat, that it must be updated from Setup mode, selecting "Update all now".

okanduzyel, 2021-03-21:
> Enabling the Auto OTA Updates option causes the firmware to search for an OTA update from the configured OTA repository every time the setup mode is selected and the setup webpage has not been loaded in a browser within 90 seconds.
> 
> If setup web page is loaded, device will restart automatically after 15 minutes, or try to update if auto OTA updates is enabled.

When It downloads and flashes itself? I have HAA v5.0.10, I can quickly open configuration page via that IP 192.168.1.38:4567, auto-update have already checked. I am confusing. 

> Sorry, I don't know, but if it's a thermostat, that it must be updated from Setup mode, selecting "Update all now".

That is sonoff basic lamb

pawelosin, 2021-03-21:
Put the device in setup mode and do nothing more. Just wait about 15 minutes.

okanduzyel, 2021-03-21:
> Put the device in setup mode and do nothing more. Just wait about 15 minutes.

Okay, It did. Thank you so much ✋🏻

closed 2021-03-21 by okanduzyel