RavenSystem/esp-homekit-devices #1371: HAA Manager does not update from 5.0.3

mrmaximas, 2021-03-22:
"Update all HAA devices" does not work
[image]
[image]

tonysprenk, 2021-03-22:
It only works from version 5.0.12. Update to latest the old fashioned way, after that you can update with HAA manager app.

xrust83, 2021-03-22:
very interesting screenshot.  explain where to read, how to set up how to use it please.

mrmaximas, 2021-03-22:
> It only works from version 5.0.12. Update to latest the old fashioned way, after that you can update with HAA manager app.

thanks!

mrmaximas, 2021-03-22:
> very interesting screenshot. explain where to read, how to set up how to use it please.

https://apps.apple.com/by/app/haa-manager/id1556105121

closed 2021-04-24 by stale[bot]

cirrusflyer, 2021-06-02:
I'm on 5.0.18 and can't get updates to work via the app.  And not sure how to do this via the web interface either.