RavenSystem/esp-homekit-devices #1373: Sonoff 4CHPROR3 Json Blinds+lightbulb

mwouda, 2021-03-24:
Hi All, 

I have a Sonoff 4CHPROR3 
I'm still strugling with the whole json code.. the .. is no help (and out of date)

I have this but the code is not accepted.. 
{
  "c": {"n":"Balkon-0FE722",
	"o": 0,
	"l": 13,
	"i": 0,
	"m": 8},
  "a": [{
	"t": 45,
	"w": 1,
	"s": 0,
	"o": 4,
	"0": {"r": [{"g": 12,"v": 1,}},"m":[{2,0}],
	"1": {"r": [{"g": 5,"v": 1,}]},"m":[{1,0}],
	"f0":[{"g": 0,"t": 1,"p": 0}],
	"f1":[{"g": 9,"t": 1,"p": 0}],
	"f2":[{ "g": 5 }, { "g": 13 }]
    },{
	"t":30,
	"g":[ CW, WW ],
	"b":[{"g":10}],}]]
}

This is how it is all connected
Blinds - Down
GPIO00	Button1 
GPIO12	Relay1
Blinds - Up
GPIO09	Button2
GPIO05	Relay2

Light Bulb ON\OFF
GPIO10	Button3
GPIO04	Relay3
Light Standby\Low\High
GPIO14	Button4
GPIO15	Relay4

Any help is appreciated. plz tell me what am i doing wrong?

mariusvasss, 2021-03-28:
"b":[{"g":10}],}]]

Corect last line.

mwouda, 2021-04-05:
Hi, 

Mariusvass: thank's for the tip. That helped some. 

I was able to get a bit further. 
 `{
    "c":{
        "n":"Balkon-0FE722",
        "o":0,
        "l":13,
        "i":0,
        "m":8
    },
    "a":[
        {
            "t":45,
            "w":1,
            "s":0,
            "o":15,
            "c":10,
			"s":1,
            "0":{"r":[{"g":12,"v":1}]},
            "1":{"r":[{"g":5,"v":1}]},
            "5":{"r":[{"g":12,"v":0}]},
			"6":{"r":[{"g":5,"v":0}]},
            "f0":[{"g":0,"t":1}],
            "f1":[{"g":9,"t":1}],
            "f2":[{"g":5},{"g":13}
            ]
        },
        {"t":30,
            "g":["W"],
            "b":[{"g":10}]
        },
		{"t":30,
            "g":["CW,WW"],
            "b":[{"g":14}],
			"f0":[{ "g": 15, "i": 0.5 }],
			"f1":[{ "g": 15, "i": 0.5 }],
			"f2":[{ "g": 15, "i": 0.5 }]
        }
    ]
}`

So i'm not able to get button4 (GPIO014) to change the state of the lamp. so for now i'm splitting them. (lack of my understanding)
button3 (GPIO010) is the on\off relay for the heater lamp. Relay3(GPIO04) [http://www.cranenbroek.nl/catalogus/images/50300-037-50020-30.jpg](url)
button4 (GPIO014) was meant to be a contact button to change the level between standby, low or high. 

I have bin able to get the window covering to work for a part. but there are a few problems. 
- The default state of the windows covers should be "open" but Home shows "closed". any idea on how to change that? The "s" does not do anything. 
- when i press button1 nothing happens. button2 seems to open and close? 
Tomorrow i'm checking to see if the Relays are correct. 

Any help is appreciated

closed 2021-05-06 by stale[bot]

jazevedodesigner, 2022-03-10:
Hello, 

Do you have the JSON CODE for window covering? For the model sonoff touch 2 gang?

 I'm putting this one,

{
"c": { "l": 0, "b": [{ "g": 2, "p": 0, "t": 5 }], "0": { "r": [{ "g": 15 }, { "g": 16 }] } },
"a": [{
"t": 45,
"o": 18,
"c": 14,
"f": 70,
"0": { "r": [{ "g": 15, "v": 1 }, { "g": 4 }] },
"1": { "r": [{ "g": 15 }, { "g": 4, "v": 1 }] },
"2": { "r": [{ "g": 15 }, { "g": 4 }] },
"3": { "r": [{ "g": 15, "i": 0.2 }, { "g": 4 }] },
"4": { "r": [{ "g": 15 }, { "g": 4, "i": 0.2 }] },
"f0": [{ "g": 5, "i": 1, "p": 0, "t":0 }],
"f1": [{ "g": 13, "i": 1, "p": 0, "t":0 }],
"f2": [{ "g": 5 }, { "g": 13 }]
}]
}

..but it doesn't work!!!!

someone can help me?