RavenSystem/esp-homekit-devices #1375: Light sensor

ooomakon, 2021-03-27:
Hello, please help me to make the light sensor work, I connected it in all known ways with different resistors, it shows 0.0001, how to make it work?

okanduzyel, 2021-03-27:
I don't know exactly but It still looks under construction, resistor value, power value etc.

https://github.com/RavenSystem/esp-homekit-devices/wiki/Light-Sensor

ooomakon, 2021-03-27:
Tried various options, doesn't show

WojtekWro, 2021-04-04:
@ooomakon 
Could you share your json?
Did you add "g": option with sensor gpio

ooomakon, 2021-04-04:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Light-Sensor

WojtekWro, 2021-04-04:
I know this example but there is missing gpio for the light sensor. That's why I asked about the "g": option.

okanduzyel, 2021-04-04:
> I know this example but there is missing gpio for the light sensor. That's why I asked about the "g": option.

- LDR is an analog device so, It must be using the analog input pin. 
- There is only one analog input. Therefore, there should not be a definition for it. 

Try to use the analog input pin. (it could be different located-named for various boards) But I don't think it is an accomplished accessory. 

[image]

ooomakon, 2021-04-04:
> Я знаю этот пример, но отсутствует gpio для датчика освещенности. Вот почему я спросил об варианте "g":.

Did you earn it?

WojtekWro, 2021-04-04:
> * LDR is an analog device so, It must be using the analog input pin.
> * There is only one analog input. Therefore, there should not be a definition for it.

Thanks @okanduzyel for your explanation.
I didn't know that. 
 I have sensor board with both analog and digital output. I use it as a contact sensor 😜. However I'm looking forward to see full support to light sensors.

closed 2021-05-06 by stale[bot]