RavenSystem/esp-homekit-devices #1378: WiFi Hotspot problem with 5.0.14.

epicsepi, 2021-03-29:
Hey! First, thank you Raven for HAA. 

My problem is, after I connect to the HAA-XXXXXX WiFi hotspot, nothing appear like before. So I cant put the ESP device to my own WiFi.

Tested on MacBook and iPhone too. Any Idea?

Thanks,
Adam

tonysprenk, 2021-03-29:
The method has changed. See updated [installation instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation).

epicsepi, 2021-03-29:
> The method has changed. See updated [installation instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation).

Very big thanks! I dont saw this new method. I hope it will works to add Home app as well. 

I have an another question. Some of tutorial videos when flashing ESP board, they flahsed with tasmota firmware first, and than HAA, with ticking the "erase the firmware". Why is this important? Now I used only fullhaaboot.bin.

Daveblanche, 2021-03-29:
If you flash with tasmota( usually a “light”version) then I found it eases the transition to uploading the larger fullhaaboot file. 
I’ve bricked a few sonoff mini (OTA) by trying to use the haa method alone, but they don’t have the memory to cope with it. Using tasmota here has smoothed the ride, for me.

epicsepi, 2021-03-29:
> If you flash with tasmota( usually a “light”version) then I found it eases the transition to uploading the larger fullhaaboot file. I’ve bricked a few sonoff mini (OTA) by trying to use the haa [...]

thank you!!

epicsepi, 2021-03-29:
> The method has changed. See updated [installation instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation).
I still have a problem. After add it to WiFi (successfully), in the Home app doesnt apper. Im just waited 30 min, plugged off and plugged in but nothing...

jiuwei000, 2021-03-30:
HAA can run perfectly on 8266, but when I use 8266-01s, I can't enter the configuration through the web page after connecting to its WiFi.

jiuwei000, 2021-03-30:
I have the same problem. HAA can run perfectly on 8266, but when I use 8266-01s, I can't enter the configuration through the web page after connecting to its WiFi.

Quimeric, 2021-04-17:
> > The method has changed. See updated [installation instructions](https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation).
> 
> I still have a problem. After add it to WiFi (successfully), in the Home app doesnt apper. Im just waited 30 min, plugged off and plugged in but nothing...

I'm have the same problem!

My Sonoff SV was updated and since that i can't find to pair again, even when i roll back a previous version.

Making some tests i removed my Sonoff Mini from the Home App and i can't find it either.

closed 2021-05-30 by stale[bot]