RavenSystem/esp-homekit-devices #1379: Flash ISSUE with MagicHome LED Controller and with NodeMCU PyFlasher

epicsepi, 2021-03-29:
[image]
[image] 

[image]

What do you think, what is the problem?

pasparakis, 2021-03-29:
I have same problem...😐

arvet333, 2021-03-31:
Hi, try to use this app
[image]

closed 2021-05-04 by stale[bot]