RavenSystem/esp-homekit-devices #1380: only W on in RGBW bulb?

toshibochan, 2021-04-02:
how to turn on only W ON on RGBW bulb or led strip when white is selected?

mariusvasss, 2021-04-02:
Make a switch just for W, a separate tile in HomeKit

toshibochan, 2021-04-02:
This is the magichome config. How I change for W as separated tile?

{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[15,13,12,14],"b":[{"g":0}]}]}

toshibochan, 2021-04-02:
And how set the button to turn ON(100%) and OFF the W color?

mariusvasss, 2021-04-02:
I'll give you a JSON tomorow

toshibochan, 2021-04-02:
Thank you @mariusvasss

mariusvasss, 2021-04-03:
{"c":{"q":850,"b":[{"g":0,"t":5}]},"a":[{"t":30,"r":5,"g":12,"v":13,"b":[{"g":0}]},{"t":30,"w":15,"b":[{"g":4,"p":0},{"g":4,"t":0}]}]}

@toshibochan 

i hope it helps u

mariusvasss, 2021-04-03:
you can use IR pads for external switch just for white

toshibochan, 2021-04-03:
@mariusvasss thank you
in "b":[{"g":4,"p":0} who the "p" is for?

mariusvasss, 2021-04-03:
I don't know. If it works, it's Ok for you. I did it a year ago...

toshibochan, 2021-04-04:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[12,5,13],"b":[{"g":0,"t":3}]},{"t":30,"g":[15],"b":[{"g":0}]}]}
for now when i press once turn ON to de last brightness but what i want is "g":[15] to press once turn ON 100% brightness, press again turn OFF.

mariusvasss, 2021-04-04:
Have you tried to make it virtual button and duble press for 100%?

mariusvasss, 2021-04-04:
Or just make it a switch

toshibochan, 2021-04-04:
I want the button as backup if the device is showing not response so I can turn on manually W color to 100% brightness.

mariusvasss, 2021-04-04:
Switch is the solution...

toshibochan, 2021-04-04:
can you make the config using this
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[12,5,13],"b":[{"g":0,"t":3}]},{"t":30,"g":[15],"b":[{"g":0}]}]}

mariusvasss, 2021-04-04:
?

toshibochan, 2021-04-04:
can you give me example

mariusvasss, 2021-04-04:
{"c":{"b":[{"g":0,"t":5}]},"a":[{"t":30,"g":[12,5,13],"b":[{"g":0,"t":3}]},{"t":1,"g":[15],"b":[{"g":0}]}]}

i think this works. just changed t30 with t1 - switch

closed 2021-04-06 by toshibochan