RavenSystem/esp-homekit-devices #1382: Lightbulb setup mode problem

okanduzyel, 2021-04-03:
Hi, 

Lightbulb, led strip whatever, all rgb lights have setup mode problem. As we know, eight time toggling it, open-close, causes to enter setup mode. RGB lights have persentage and color options. Every actions, such as percentage changing, color changing, increase setup mode count which is eight and device, unfortunately, enters unexpectedly setup mode. 

Try to tamper your rgb light a little, you will see that after a short time, it will turn "no response" and open setup page...

toshibochan, 2021-04-03:
Somehow if you can add little button to gpio 0 you can just hold for 8 second

okanduzyel, 2021-04-03:
> Somehow if you can add little button to gpio 0 you can just hold for 8 second 

I don't want that it enters setup mode. It enters unexpectedly when I use it. For example, changing colors, light levels etc. clear?

toshibochan, 2021-04-03:
All depends how you do the json code. What is your config?

toshibochan, 2021-04-03:
This part is the 8s hold "b":[{"g":0,"t":5}]

toshibochan, 2021-04-03:
And usually we don’t hold button for that long for turn on the light

okanduzyel, 2021-04-03:
```
{
  "c":{"q":500,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":30,
    "g":[15,13,12],
    "b":[{"g":0}]
  }]
}
```

I say samething 3rd times... 

> This part is the 8s hold "b":[{"g":0,"t":5}]

This is physical way to enter setup mode. On the other hand, there is another way to enter setup mode that open-close (toggle) the accessory on homekit 8 times. 

For RGb lights, the problem is that, 

Not only open-close but also, Chaning color, changing light level increases the setup mode counting. When you try to change color, persentage of light level etc. It enters unexpectedly in setup mode and turns "no response".

Now, is it clear?

toshibochan, 2021-04-03:
Unfortunately toggle to enter setup mode is in the code and I think cannot be changed in json config

tonysprenk, 2021-04-03:
With the json you posted setup mode should only be invoked when gpio 0 is pulled high. The only logical explanation for setup mode being invoked when you control the lights is that gpio0 is pulled high for 8 seconds when you do so. I say this even though I can see in the json example you posted (is that the actual json you’re using?) nothing except for the physical button is connected to gpio0. So that makes the behaviour you’re describing confusing.

tonysprenk, 2021-04-03:
In other words: setup mode shouldn’t be invoked by pulling any gpio high except for the one specified.

okanduzyel, 2021-04-03:
4th time I am explaining...

**"ANY ACCESSORY"** open-close it on software, It will enter the setup mode. -> https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode

> **To enter setup mode, you must quickly toggle any action switch/button 8 times (physical or from Home App). You can secure this method by declaring a time after the device boot to perform it.**
> 
> Additionally, you can declare a physical method to enter setup mode.
> 
> If the device's configured WiFi network is unavailable when entering setup mode it will switch to AP mode and you will need to connect directly to it via the generated SSID e.g. HAA-XXXXXX (where XXXXXX are the last six characters of its Ethernet MAC address), and then open URL in a web browser:
> 
> http://192.168.4.1:4567
> Emergency Setup Mode: If a device has its power cut or freezes within 2 seconds of boot, the next time it boots it will go directly to setup mode.

Turn on RGB bulb 1 action,
Change color 1 action,
Change color 1 action,
Change color 1 action,
Change the percentage of light level 1 action,
Change color 1 action,
Change color 1 action,
Change the percentage of light level 1 action,

Devices enter setup mode and turn no response. This is not a physical thing, gpio 0 button problem, etc. This is described in code. 

I say that the problem is every action increases the setup mode entering count.

Turn on RGB bulb 1 action,
**Change color 1 action,
Change color 1 action,
Change color 1 action,
Change percentage of light level 1 action,
Change color 1 action,
Change color 1 action,
Change percentage of light level 1 action,**

These I marked above, shouldn't increase the setup mode counting. Therefore, I will not enter setup mode while you try to change color, percentage of light level etc.

Before commenting, please try it on board, I am trying it on Lolin V3 board, and observe what I say.

tonysprenk, 2021-04-03:
Can you not just use setup mode timer?

https://github.com/RavenSystem/esp-homekit-devices/wiki/General-Configuration#Setup-Mode-Timer

okanduzyel, 2021-04-03:
It is not about setup timer, gpio0, or any other else.

tonysprenk, 2021-04-03:
But with setup mode timer your problem will be solved. Because you will not enable setup mode accidentally after the timer is finished.

okanduzyel, 2021-04-03:
@tonysprenk Here is the **issue** side of HAA, not forum, problem solving place etc. I comment its issue to inform @RavenSystem. It should not work like that.

tonysprenk, 2021-04-03:
I’m arguing that it should work like that (because that’s the way it is) and if you don’t want to use it that way you can disable it by using the setup mode timer.

tonysprenk, 2021-04-03:
What you’re describing is not an issue in my opinion because the behaviour you’re describing is to be expected, as explained in the wiki. If you would rather not see this behaviour happening then it would be more like a feature request. All though I don’t think @RavenSystem will do anything with it because he has already put in place a system for disabling the behaviour you’re describing, which is the setup mode timer.

okanduzyel, 2021-04-03:
It should enter setup mode while open-close (on homekit)
It should **NOT** enter setup mode while changing lamb color or changing the percentage of the light level.

now the issue must be crystal clear.

tonysprenk, 2021-04-03:
You’re issue is clear. You can wait to solve your issue on @RavenSystem to rewrite his code (which he most likely will not do because he has already made a solution for you, setup mode timer) or you can solve your issue now, by using setup mode timer. Obviously when using setup mode timer you won’t be able to invoke setup mode after a certain time after boot so the solution is not perfect. But this too can be solved with a separate virtual setup mode button in json, emergency setup mode, or using the HAA-manager app to invoke setup mode.

toshibochan, 2021-04-03:
Be cool if have a switch showing in EVE app like this https://github.com/maccoylton/esp-homekit-rgbw-strip
[image]

tonysprenk, 2021-04-03:
@toshibochan You can add a dummy switch to your json that activates setup mode. Just add: `{"1":{"s":[{"a":1}]},"i":1} ` to the end of your json. I don’t use the eve app so I’m not sure how it will show up but in the home app it creates a separate toggle switch to invoke setup mode.

toshibochan, 2021-04-03:
@tonysprenk thanks I will try later

RavenSystem, 2021-04-03:
Nothing to do here. In lightbulb accessory, setup trigger counts are not affected when color or brightness changes. I tested it with this JSON just now (HAA v5.0.14):
```
{
  "c":{"q":500,"b":[{"g":0,"t":5}]},
  "a":[{
    "t":30,
    "g":[15,13,12],
    "b":[{"g":0}]
  }]
}
```

Setup mode is not called changing brightness or color; only toggling between on and off.

okanduzyel, 2021-04-03:
https://user-images.githubusercontent.com/38145537/113490797-2e42b280-94d5-11eb-91c3-e1803c2ef9c6.mov

[image]

So what is this? I can do it?

RavenSystem, 2021-04-03:
My test is right:
https://user-images.githubusercontent.com/8655274/113491057-3814e780-94ce-11eb-9e5f-cfe6afadec29.mp4

Logs are needed to know what is happening.

okanduzyel, 2021-04-03:
> My test is right:
> https://user-images.githubusercontent.com/8655274/113491057-3814e780-94ce-11eb-9e5f-cfe6afadec29.mp4
> 
> Logs are needed to know what is happening.

```
Jan 01 00:00:11 >>> mDNS_announce: sent
Jan 01 00:00:15 >>> mDNS_announce: working with TTL 4500
Jan 01 00:00:15 >>> mDNS_announce: sent
Jan 01 00:00:18 [4] New 192.168.1.39:2804 (1/8)
Jan 01 00:00:18 [4] Verify 1/2
Jan 01 00:00:18 [4] Verify 2/2
Jan 01 00:00:18 [4] Found pairing with 6691761C-BD85-4685-A1C3-750F90F7A4DD
Jan 01 00:00:18 [4] Verify OK
Jan 01 00:00:18 [4] Get Acc
Jan 01 00:00:18 [4] Get Ch
Jan 01 00:00:19 [4] Update Ch
Jan 01 00:00:24 NTP update (-1)
Jan 01 00:00:30 [4] Update Ch
Jan 01 00:00:30 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:30 <1> Lightbulb setter HUE 0, SAT 0
Jan 01 00:00:30 <1> Target Color = 65535, 53158, 46309, 65535, 0
Jan 01 00:00:30 <1> Exec Action 1
Jan 01 00:00:30 <1> Wildcard 0 100
Jan 01 00:00:31 Color fixed
Jan 01 00:00:32 [4] Update Ch
Jan 01 00:00:32 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:32 <1> Lightbulb setter HUE 1, SAT 100
Jan 01 00:00:32 New chromaticity: [0.690704, 0.308213]
Jan 01 00:00:32 <1> Target Color = 65535, 83, 0, 0, 0
Jan 01 00:00:32 <1> Exec Action 1
Jan 01 00:00:32 <1> Wildcard 0 100
Jan 01 00:00:32 [4] Update Ch
Jan 01 00:00:32 [4] Update Ch
Jan 01 00:00:32 [4] Update Ch
Jan 01 00:00:32 Color fixed
Jan 01 00:00:33 [4] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 287, SAT 49
Jan 01 00:00:33 New chromaticity: [0.376739, 0.243516]
Jan 01 00:00:33 <1> Target Color = 65535, 3568, 64482, 64482, 0
Jan 01 00:00:33 <1> Exec Action 1
Jan 01 00:00:33 <1> Wildcard 0 100
Jan 01 00:00:33 [4] Update Ch
Jan 01 00:00:33 [4] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 190, SAT 100
Jan 01 00:00:33 New chromaticity: [0.138349, 0.277025]
Jan 01 00:00:33 <1> Target Color = 0, 32538, 65535, 0, 0
Jan 01 00:00:33 <1> Exec Action 1
Jan 01 00:00:33 <1> Wildcard 0 100
Jan 01 00:00:33 [4] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 98, SAT 100
Jan 01 00:00:33 New chromaticity: [0.204656, 0.665877]
Jan 01 00:00:33 <1> Target Color = 8018, 65535, 0, 0, 0
Jan 01 00:00:33 <1> Exec Action 1
Jan 01 00:00:33 <1> Wildcard 0 100
Jan 01 00:00:34 [4] Update Ch
Jan 01 00:00:34 [4] Update Ch
Jan 01 00:00:34 [4] Update Ch
Jan 01 00:00:34 Color fixed
Jan 01 00:00:34 [4] Update Ch
Jan 01 00:00:34 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:34 <1> Lightbulb setter HUE 41, SAT 100
Jan 01 00:00:34 New chromaticity: [0.562231, 0.402751]
Jan 01 00:00:34 <1> Target Color = 65535, 20293, 0, 0, 0
Jan 01 00:00:34 [4] Update Ch
Jan 01 00:00:34 <1> Exec Action 1
Jan 01 00:00:34 <1> Wildcard 0 100
[... 128 lines cut ...]
Jan 01 00:00:43 [4] Update Ch
Jan 01 00:00:44 [4] Update Ch
Jan 01 00:00:44 [4] Update Ch
Jan 01 00:00:44 <1> Lightbulb setter ON 1, BRI 100
Jan 01 00:00:44 <1> Lightbulb setter HUE 98, SAT 0
Jan 01 00:00:44 <1> Target Color = 65535, 53158, 46309, 65535, 0
Jan 01 00:00:44 <1> Exec Action 1
Jan 01 00:00:44 <1> Wildcard 0 100
Jan 01 00:00:44 [4] Update Ch
Jan 01 00:00:45 Color fixed
Jan 01 00:00:45 <1> Exec Action 1
Jan 01 00:00:45 <1> Wildcard 0 100
Jan 01 00:00:45 Color fixed
Jan 01 00:00:45 [4] Update Ch
Jan 01 00:00:45 Setup mode call

Jan 01 00:00:45 
Jan 01 00:00:45 Rebooting...
Jan 01 00:00:45 rm match
Jan 01 00:00:45 del if0
Jan 01 00:00:45 usl
Jan 01 00:00:45 sul 0 0
Jan 01 00:00:45 ! [4] Reading from socket (113). Closing
Jan 01 00:00:45 [4] Closed (0/8)
```

RavenSystem, 2021-04-03:
Please, update to 5.0.15 and capture logs again.

okanduzyel, 2021-04-03:
Now, HAA 5.0.15, I could never put it in the setup mode when It opened with changing color or percentage. The problem looks is solved but I found a new issue. 

When its percentage is %0 (closed), touch the color buttons, my bulb is flickering when I touch the colors, after a while, you will see that it opens the setup page again. 

https://user-images.githubusercontent.com/38145537/113492886-a87a3380-94e3-11eb-84b5-53bcde447bd7.mov

[image]

toshibochan, 2021-04-04:
> @toshibochan You can add a dummy switch to your json that activates setup mode. Just add: `{"1":{"s":[{"a":1}]},"i":1} ` to the end of your json. I don’t use the eve app so I’m not sure how it will show up but in the home app it creates a separate toggle switch to invoke setup mode.

it work but i'm afraid to add this switch and Siri turn on all my device to setup mode.

okanduzyel, 2021-04-05:
@RavenSystem the last issue is still here with HAA 5.0.16

okanduzyel, 2021-04-05:
```
Jan 01 00:00:22 NTP update (-1)
Jan 01 00:00:25 [6] Update Ch
Jan 01 00:00:25 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:25 <1> Lightbulb setter HUE 1, SAT 100
Jan 01 00:00:25 Setup mode trigger 1/8
Jan 01 00:00:25 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:25 <1> Exec Action 0
Jan 01 00:00:26 [7] New 192.168.1.47:62936 (4/8)
Jan 01 00:00:26 [7] Verify 1/2
Jan 01 00:00:26 [7] Verify 2/2
Jan 01 00:00:26 [7] Found pairing with D25D8C1C-0483-482A-9AC5-7D395A6C3F27
Jan 01 00:00:26 [7] Verify OK
Jan 01 00:00:27 [7] Get Acc
Jan 01 00:00:29 [6] Update Ch
Jan 01 00:00:30 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:30 <1> Lightbulb setter HUE 98, SAT 100
Jan 01 00:00:30 Setup mode trigger 1/8
Jan 01 00:00:30 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:30 <1> Exec Action 0
Jan 01 00:00:30 [6] Update Ch
Jan 01 00:00:30 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:30 <1> Lightbulb setter HUE 190, SAT 100
Jan 01 00:00:30 Setup mode trigger 2/8
Jan 01 00:00:30 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:30 <1> Exec Action 0
Jan 01 00:00:31 [6] Update Ch
Jan 01 00:00:31 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:31 <1> Lightbulb setter HUE 41, SAT 100
Jan 01 00:00:31 Setup mode trigger 3/8
Jan 01 00:00:31 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:31 <1> Exec Action 0
Jan 01 00:00:31 [6] Update Ch
Jan 01 00:00:31 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:31 <1> Lightbulb setter HUE 222, SAT 18
Jan 01 00:00:31 Setup mode trigger 4/8
Jan 01 00:00:31 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:31 <1> Exec Action 0
Jan 01 00:00:32 [6] Update Ch
Jan 01 00:00:32 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:32 <1> Lightbulb setter HUE 0, SAT 0
Jan 01 00:00:32 Setup mode trigger 5/8
Jan 01 00:00:32 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:32 <1> Exec Action 0
Jan 01 00:00:32 [6] Update Ch
Jan 01 00:00:32 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:32 <1> Lightbulb setter HUE 1, SAT 100
Jan 01 00:00:32 Setup mode trigger 6/8
Jan 01 00:00:32 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:32 <1> Exec Action 0
Jan 01 00:00:32 [6] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 98, SAT 100
Jan 01 00:00:33 Setup mode trigger 7/8
Jan 01 00:00:33 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:33 <1> Exec Action 0
Jan 01 00:00:33 [6] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 190, SAT 100
Jan 01 00:00:33 Setup mode trigger 8/8
Jan 01 00:00:33 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:33 <1> Exec Action 0
Jan 01 00:00:33 [6] Update Ch
Jan 01 00:00:33 <1> Lightbulb setter ON 0, BRI 100
Jan 01 00:00:33 <1> Lightbulb setter HUE 41, SAT 100
Jan 01 00:00:33 Setup mode trigger 9/8
Jan 01 00:00:33 <1> Target Color = 0, 0, 0, 0, 0
Jan 01 00:00:33 <1> Exec Action 0
Jan 01 00:00:34 [6] Update Ch
Jan 01 00:00:34 <1> Lightbulb setter ON 0, BRI 100
[... 102 lines cut ...]
Saved BSSID: 4c9eff4bb45c
Wifi Mode: Forced BSSID
HAA Setup
Enabling auto reboot
Hostname: HAA-050976
mode : sta(cc:50:e3:05:09:76) + softAP(ce:50:e3:05:09:76)
add if1
bcn 100
Function called without core lock
Start AP SSID=HAA-050976
bcn 0
del if1
add if1
bcn 100
Start DHCP server
Start HTTP server
scandone
reconnect
scandone
add 0
aid 6
cnt 
```

Lamp is closed, just touched the colors.

okanduzyel, 2021-04-17:
The problem is still continue with HAA v5.0.17 @RavenSystem 
I am closing this issue.

closed 2021-04-17 by okanduzyel

okanduzyel, 2021-04-24:
It is still here, despite updating HAA v5.0.18.  @RavenSystem