RavenSystem/esp-homekit-devices #1386: New wifi network

orson32, 2021-04-05:
I changed the name and password of my network, how can I reconfigure the devices for the new network?

toshibochan, 2021-04-05:
You have to go to setup mode and go to you browse device IP:4567

orson32, 2021-04-06:
> You have to go to setup mode and go to you browse device IP:4567

how do i go to setup mode? And what is url?

toshibochan, 2021-04-06:
If you have a button connect to gpio 0, hold for 8 sec or more and release, go to the browser. Use device IP address. Example 192.168.1.67:4567

toshibochan, 2021-04-06:
Or https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode

orson32, 2021-04-06:
> If you have a button connect to gpio 0, hold for 8 sec or more and release, go to the browser. Use device IP address. Example 192.168.1.67:4567

Excelent! Thanks so much

closed 2021-04-06 by orson32

xrust83, 2021-04-06:
https://youtu.be/EjFnVP9Jsj8