RavenSystem/esp-homekit-devices #1391: TV Service only show power on/off and source

xyhh, 2021-04-07:
[image]

toshibochan, 2021-04-07:
In the setting you have to select show volume as dimmer
[image]

xrust83, 2021-04-07:
And what relationship does your homebridge plugin have if I'm not mistaken "lg webostv" to the HAA Remote device that is asked about above?

toshibochan, 2021-04-07:
Sorry was mistake

tonysprenk, 2021-04-07:
The “remote” in the home app only shows sources so you can setup automations with them. The remote in control centre should show a remote with all buttons. Once your in the remote in control centre you can use the physical volume buttons of your iPhone to control the volume of your tv.

closed 2021-05-06 by RavenSystem