RavenSystem/esp-homekit-devices #1393: Cannot Reset Homekit ID

skupu, 2021-04-08:
Hi,

I flashed Gruppio with HAA and cannot add device to HomeKit - it says the accessoty is already existing or the code is not correct or just cannot connect (timeout).
Erasing flash, reflashing, putting different JSON configs, restarting iOS devices, restarting router do not help.

What to do?

skupu, 2021-04-08:
@RavenSystem Tasmota works correctly, Sonoff works, just HAA has problem...
Maybe this is a Homekit issue?

skupu, 2021-04-09:
I already fixed it myself. Ask if you want to know how.

Quimeric, 2021-04-16:
After update to the latest fullhaaboot (5.0.16) my Sonoff SV do not appear to be paired.

I've reflashed several times, changed the JSON and WIFI but the accessory it never shows up to be paired.

I think it's the same problem that you've faced, how do you solved?

Daveblanche, 2021-04-16:
Have you reset the HomeKit id?
I got caught out with that, just once. Never again.

Quimeric, 2021-04-16:
When i reflash the device do not show the reset box but i try once or twice.

I’m having difficulty to enter in setup mode.

Enviado do meu iPhone

skupu, 2021-04-16:
So my solution, which aparently needs to be little bit improved (as it requires a few attempts) is to:
1. create new wifi with internet access
2. use the 8 times/few seconds button/switch to go into wifi setup
3. connect to new wifi
4. go to setup from another device connected to the created wifi
5. put the update details to some older release (i used 5.0.12) or just the newest
6. wait untill system updates
7. reset again
8. maybe again
9. maybe repeat previous steps (2-7) until you get result
10. finally the homekit reset will appear
11. follow the standard way to configure the desired wifi and json config
12. have fun with device on your homekit

Daveblanche, 2021-04-16:
I may be repeating here, sorry if I am. I’m not at my computer. 
If it still remembers the network, and shows on your router with an up address, then go to that address on a browser and add :4567
So..  10.0.10.34:4567.  For example. 
I thought mine weren’t going into setup mode for a while as they didn’t fire up their hotspot when the software changed a while ago.  So I went through something similar. Hope it helps.

Quimeric, 2021-04-17:
> So my solution, which aparently needs to be little bit improved (as it requires a few attempts) is to:
>
> 1. create new wifi with internet access
> 2. use the 8 times/few seconds button/switch to go [...]

How do i custom update? I read about but i don't understand hahaha

The new wifi you said is a hotspot?

Quimeric, 2021-04-17:
> I may be repeating here, sorry if I am. I’m not at my computer. If it still remembers the network, and shows on your router with an up address, then go to that address on a browser and add :4567 [...]

In the new fullhaaboot.bin, the 5.0.16 version, i've to put the http://192.168.4.1:4567 to configure it.

But my difficulty to enter in setup mode is when i've setup de wifi. The Sonoff SV button won't enable setup mode i don't know why, even when i press it 8 times, nor holding down the button for awhile.

So, i always try the emergency boot and sometimes i can't connect to de sonoff hotspot or access via http://haa-xxxx:4567.

What am i doing wrong?

Daveblanche, 2021-04-17:
The generic IP address, 192.168.4.1:4567 won’t work if it is already set up on WiFi for your system: in my limited experience. 
This is when you have to try to use the IP address it was assigned by your router, using dhcp or reserve or static ip, when you set it up.

So if your router gave it 192.168.1.23, for example, you would enter that IP address with “:4567” after it.

I never used the haaxxxx web address as my Mac seems to dislike it so can’t comment there.

skupu, 2021-04-17:
Exactly. If you connect directly to the device, use the 4.1:4567.
If it already connects the Hotspot, you can check the IP with Net Analyzer and go to the IP:4567 to reach the interface.

The custom update is easy: in the first field put: github.com/RavenSystem/haa/releases/download/5.0.12

(or another release, like 5.0.16) and below:

443

Pay attention that if you scan for wifi the checkboxes and fields are being cleared/resetted.

toshibochan, 2021-04-17:
if is new device and is not connected to your home WiFi:
you have to connect you computer or phone to the device WiFi. After you connect the computer or phone to the device WiFi use the browser 192.168.4.1:4567

if you have setup the wifi and add to homekit Look serial number in home app. Serial example ABCDEG-1.
HAA-ABCDEG.local:4567 in your browser. If your device have button, turn on the device(light), hold the button For about 10 sec and release. the device will turn off. Wait a bit. Go to browser and use HAA-ABCDEG.local:4567.

if don’t have button and connected to home WiFi and is in home app.
Use the home app and turn ON-OFF-ON-OFF-ON-OFF-ON-OFF total 8
Wait a bit use browser HAA-ABCDEG.local:4567

Daveblanche, 2021-04-17:
But to add to this
The HAA manager app is an absolute game changer!
No more multiple presses, or dummy switches (for me) or trying to press the button for 8 secs.
Awesome...and simple, although I bet it took lots of work to develop.

mariusvasss, 2021-04-17:
I have the app too! I can update all devices at once from all homes.

skupu, 2021-04-17:
My devices are paired with home assitant and haa doesn't work

xrust83, 2021-04-17:
https://youtu.be/EjFnVP9Jsj8

I will try to make subtitles in English.  but not today ;-)

skupu, 2021-04-17:
This tutorial is very helpful!
Thank you!
Of course subtitles will help a lot!

Quimeric, 2021-04-17:
> The generic IP address, 192.168.4.1:4567 won’t work if it is already set up on WiFi for your system: in my limited experience. This is when you have to try to use the IP address it was assigned by [...]

Yes! The haa-xxxx don't work, im using now the ip and the setup works.

Quimeric, 2021-04-17:
> Exactly. If you connect directly to the device, use the 4.1:4567.
> If it already connects the Hotspot, you can check the IP with Net Analyzer and go to the IP:4567 to reach the interface.
>
> The custom [...]

I need to check the OTA Updates or the Update All Now in this case? I'll definitely try it on my other device too.

Quimeric, 2021-04-17:
> But to add to this The HAA manager app is an absolute game changer! No more multiple presses, or dummy switches (for me) or trying to press the button for 8 secs. Awesome...and simple, although I bet it took lots of work to develop.
> […](#)

I have to say, i couldn't use via HAA Manager.

Before the Sonoff SV stop working, it was my first device to be updated to 5.0.16 and be supported by the app.

I try to enter on setup mode via app and nothing happen.

I don't know what I've could possibly done wrong.

Anyway, my struggle now is to make my Sonoff SV working as a garage door accessory to be found in the home app search.

Quimeric, 2021-04-17:
> github.com/RavenSystem/haa/releases/download/5.0.12

Update about that procedure:

In my Sonoff SV (which i already updated to the latest firmware and back to older for test) i was able to update, I've put the "github.com/RavenSystem/haa/releases/download/5.0.17" on custom server and "443" on port, checked the Auto OTA and the Update All Now and saved. Almost instantly the 5.0.17 showed in the setup.

In my Sonoff Mini with firmware 3.5.1 i do the same and the firmware won't update. :(

skupu, 2021-04-17:
You may just need to flash it again with haaboot.bin

Quimeric, 2021-04-17:
> You may just need to flash it again with haaboot.bin

Instead of flash fullhaaboot.bin wiping data i just flash haaboot.bin without wipe data?

Daveblanche, 2021-04-17:
Marvellous!

Quimeric, 2021-04-20:
Update: any iGadgets can find the Sonoff SV or Mini accessories.

I've opened the issue: https://github.com/RavenSystem/esp-homekit-devices/issues/1408

After several firmware upgrades and downgrades and JSON changes the Sonoff looks fine but simply won't show up.

mariusvasss, 2021-04-20:
If it's not the latest firmware, wait about 10 minutes to update after Wi-Fi config.

Quimeric, 2021-04-20:
Now i removed the home and start a new one.

Nothing can be add.

Hoobs, Sonoff TX, Mini, SV, nothing appears.

When i scan the QR codes it enter in a eternal configuration until "Accessory not found".

Looks like a iOS/iPadOS problem...

skupu, 2021-04-23:
You need to reset homekit ID in all your devices to add them to your new home.

closed 2021-06-04 by stale[bot]