RavenSystem/esp-homekit-devices #1395: BL602 support?

mariusvasss, 2021-04-09:
Magichome comes with this chip. I've got 3 and I can't flash them.

urbaniak, 2021-04-09:
the same bl602 as https://pine64.com/product/pinecone-bl602-evaluation-board/?v=0446c16e2e66?

that's totally unrelated to esp, it's a different chip with different architecture (risc-v vs xtensa)

there are some work on arduino framework support for bl602 here for example https://github.com/pine64/ArduinoCore-bouffalo, official sdk https://github.com/bouffalolab/bl_iot_sdk, but we're quite far from getting homekit support there

mariusvasss, 2021-04-09:
Thanks. Just asking to know if it is on to do list. Look's like this chip will be the new ESP....
Thank you again.

okanduzyel, 2021-04-11:
I wonder which value of clock speed that used in HAA? esp8266 default clock frequency is 80Mhz and it can reach up to 160Mhz. I think that esp8266 is a very slow responsive device compare with esp32 which is ten times faster than esp8266. (I compare with these because homekit runs on  them)

If 80mhz used, bl602 is more than two times faster device.

mariusvasss, 2021-04-11:
And BT, and much more...

closed 2021-05-30 by stale[bot]