RavenSystem/esp-homekit-devices #1396: Config for firmware update?

toshibochan, 2021-04-10:
Have dummy switch example for firmware update?

okanduzyel, 2021-04-11:
```
  {
    "i":0.5,
    "1":{ "s":[{ "a":1 }]}
  }
```

toshibochan, 2021-04-11:
This will enter in firmware update without need to go setup mode?

okanduzyel, 2021-04-12:
I don't think so. It opens setup mode and I guess you need to enter setup mode via IP address like 192.xxxxxxxx:4567.

toshibochan, 2021-04-12:
I wish I could do firmware update without go to web interface.

WojtekWro, 2021-04-14:
{
    "sn":"UPD",
    "1":{"s":[{"a":2}]}
    },
    {
    "sn":"SET",
    "1":{"s":[{"a":1}]}
    }

WojtekWro, 2021-04-14:
The first one updates firmware without entering setup mode. 
The second one is entering setup mode.

toshibochan, 2021-04-15:
@WojtekWro it works thanks thanks thanks

toshibochan, 2021-04-15:
have way to show device ip address in HomeKit?

Quimeric, 2021-04-17:
> {
> "sn":"UPD",
> "1":{"s":[{"a":2}]}
> },
> {
> "sn":"SET",
> "1":{"s":[{"a":1}]}
> }

I have some devices with old firmware (2.3.2 and 3.5.1) but i never done OTA update and i don't know exactly how. 

Just adding the first button will do the job?

closed 2021-05-30 by stale[bot]