RavenSystem/esp-homekit-devices #1401: Inter-command Delay for UART Action not work - help

PugachevDB, 2021-04-15:
Device: LC 5V 2-Channel WiFi Relay X2
2 Channels Relay

JSON:
{"c":{"b":[{"g":0,"t":4}],"r":[{"n":0, "s":115200, "p":0, "b":0}]},
"a":[{"t":40,"d":14,
"0" : {"u":
[{"n":0,"d":1000,"v":"A00101A2"},
 {"n":0,"v":"A00100A1"}]},
"1":{"u":
[{"n":0,"d":1000,"v":"A00201A3"},  
 {"n":0,"v":"A00200A2"}]},
"2":{"a":0},
"3":{"a":1}}]}

And the result is (i could see it in coolterm app): 
When press OPEN - A00101A2A00100A1 - without pause or delay
CLOSE - A00201A3A00200A2 - without pause or delay
What's wrong? Where is my fault? Help me, please

RavenSystem, 2021-04-15:
I can confirm it's is a bug. It will be fixed in next HAA release.

PugachevDB, 2021-04-16:
Thanks a lot

_____________________

Yours faithfully,
Pugachev Dmitriy

RavenSystem, 2021-04-16:
Try v5.0.17.

PugachevDB, 2021-04-17:
Perfect, thanks

_____________________

Yours faithfully,
Pugachev Dmitriy

closed 2021-04-17 by RavenSystem