RavenSystem/esp-homekit-devices #1402: DHT11 module + ESP-01S

andrewbpowers, 2021-04-16:
Hello!

I try to connect a module with a DHT11 (temperature and humidity sensor) and a ESP-01S, but it doesn't work.
This is the JSON Config I use: { "c":{ "sn":"ESP01-1", "n":"ESP01-1" }, "a":[ { "t":24, "g":2, "n":1 } ] }

Does anyone have a solution or an idea what the problem could be?
Thank you very much!

komarofleo, 2021-04-16:
Hi, use this 
{ "c":{ "sn":"ESP01-1", "n":"ESP01-1" }, "a":[ { "t":24, "g":2, "n":1 } ] }
and try 3.3V power supply!

andrewbpowers, 2021-04-16:
Thank you very much!
Ok, I erase everything with esptool.py and flashing the fullhaaboot.bin.
But, after some minutes "Connecting to Sensor"... again "Unable to Add Accessory". :-(

andrewbpowers, 2021-04-16:
I restart the router, the sensor module with the ESP-01S and the iPhone. After that, it was possible to add... crazy. But I seems it was the typically Homekit "no response" error thing, which is really annoying. 

Thank you very much for your help!

closed 2021-05-30 by stale[bot]