RavenSystem/esp-homekit-devices #1403: UART 2 not usable

ChristianH99, 2021-04-17:
Whenever I use UART 2 in an accessory config the JSON isn't accepted anymore. In the general config part it doesn't cause any problems. After saving the configuration the device stay in setup mode and can't be added to the home app.

Example config:
`{
  "c":{
    "sn":"NodeMCU-1",
    "n":"NodeMCU-1",
    "o":1,
    "m":60,
    "r":[
      {
        "n":2,
        "s":115200
      }
    ]
  },
  "a":[
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":2,
            "v":"00"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":2,
            "v":"01"
          }
        ]
      }
    }
  ]
}`

RavenSystem, 2021-04-17:
There is not UART2, only UART0 and UART1. It seem that you are using UART0, because you declare: `"r":[ { "n":2, "s":115200 } ]`. Then, your JSON should be:
`{ "c":{ "sn":"NodeMCU-1", "n":"NodeMCU-1", "o":1, "m":60, "r":[ { "n":2, "s":115200 } ] }, "a":[ { "t":1, "s":5, "0":{ "u":[ { "v":"00" } ] }, "1":{ "u":[ { "v":"01" } ] } } ] }`

ChristianH99, 2021-04-17:
Aaah, I understand, thank you. I thought I had to declare "n";2 in the actions as well, because of this sentence in the wiki "The UART Number defines the UART port to use. See UART Configuration for details of port and settings."

RavenSystem, 2021-04-17:
I have modified Wiki to try to clarify:

https://github.com/RavenSystem/esp-homekit-devices/wiki/Accessory-Configuration#UART-Actions

ChristianH99, 2021-04-17:
Sounds good, perfekt. Thanks for your work!

closed 2021-04-17 by ChristianH99