RavenSystem/esp-homekit-devices #1406: Fatal exception when using one less accessory

ChristianH99, 2021-04-18:
I have a project including 12 switches and 2 leak sensors. During testing I had one more switch in my JSON and this is working just fine. 
JSON: `{"c":{"sn":"NodeMCU-1","n":"NodeMCU-1","o":1,"m":60,"r":[{"n":1,"s":115200,"p":0,"b":0}]},"a":[{"t":2,"s":5,"cd":5},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"00"}]},"1":{"u":[{"n":1,"v":"01"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"10"}]},"1":{"u":[{"n":1,"v":"11"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"20"}]},"1":{"u":[{"n":1,"v":"21"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"30"}]},"1":{"u":[{"n":1,"v":"31"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"40"}]},"1":{"u":[{"n":1,"v":"41"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"50"}]},"1":{"u":[{"n":1,"v":"51"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"60"}]},"1":{"u":[{"n":1,"v":"61"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"70"}]},"1":{"u":[{"n":1,"v":"71"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"80"}]},"1":{"u":[{"n":1,"v":"81"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"90"}]},"1":{"u":[{"n":1,"v":"91"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"A0"}]},"1":{"u":[{"n":1,"v":"A1"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"B0"}]},"1":{"u":[{"n":1,"v":"B1"}]}},{"t":7,"f0":[{"g":16,"t":0}],"f1":[{"g":16,"t":1}]},{"t":7,"f0":[{"g":5,"t":0}],"f1":[{"g":5,"t":1}]}]}`

When I leave out the first accessory in the list I get a fatal exception when booting and get back into setup mode. 
JSON: `{"c":{"sn":"NodeMCU-1","n":"NodeMCU-1","o":1,"m":60,"r":[{"n":1,"s":115200,"p":0,"b":0}]},"a":[{"t":1,"s":5,"0":{"u":[{"n":1,"v":"00"}]},"1":{"u":[{"n":1,"v":"01"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"10"}]},"1":{"u":[{"n":1,"v":"11"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"20"}]},"1":{"u":[{"n":1,"v":"21"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"30"}]},"1":{"u":[{"n":1,"v":"31"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"40"}]},"1":{"u":[{"n":1,"v":"41"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"50"}]},"1":{"u":[{"n":1,"v":"51"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"60"}]},"1":{"u":[{"n":1,"v":"61"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"70"}]},"1":{"u":[{"n":1,"v":"71"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"80"}]},"1":{"u":[{"n":1,"v":"81"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"90"}]},"1":{"u":[{"n":1,"v":"91"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"A0"}]},"1":{"u":[{"n":1,"v":"A1"}]}},{"t":1,"s":5,"0":{"u":[{"n":1,"v":"B0"}]},"1":{"u":[{"n":1,"v":"B1"}]}},{"t":7,"f0":[{"g":16,"t":0}],"f1":[{"g":16,"t":1}]},{"t":7,"f0":[{"g":5,"t":0}],"f1":[{"g":5,"t":1}]}]}`
Although I notices that heap is running low before the last accessory is initialized.

Serial log is attached.
[Log file with error.txt](https://github.com/RavenSystem/esp-homekit-devices/files/6331399/Log.file.with.error.txt)

ChristianH99, 2021-04-18:
Seems like this is a timing problem. I ran into another issue with more than nine accessories: The creation of the HomeKit ID and key and the mDNS announcement wasn't done at all and therefore I wasn't able to add the device to the HomeApp.

After adding a few "cd":10 to some accessories above JSON I had no more problems. The fatal exception is gone and HomeKit announcement is sent properly.

JSON with delays: `{
  "c":{
    "sn":"NodeMCU-1",
    "n":"NodeMCU-1",
    "o":1,
    "m":60,
    "r":[
      {
        "n":1,
        "s":115200,
        "p":0,
        "b":0
      }
    ]
  },
  "a":[
    {
      "t":1,
      "s":5,
      "cd":10,
      "0":{
        "u":[
          {
            "n":1,
            "v":"00"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"01"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"10"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"11"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"20"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"21"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"30"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"31"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"40"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"41"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"50"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"51"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"60"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"61"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"70"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"71"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"80"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"81"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "0":{
        "u":[
          {
            "n":1,
            "v":"90"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"91"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "cd":10,
      "0":{
        "u":[
          {
            "n":1,
            "v":"A0"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"A1"
          }
        ]
      }
    },
    {
      "t":1,
      "s":5,
      "cd":10,
      "0":{
        "u":[
          {
            "n":1,
            "v":"B0"
          }
        ]
      },
      "1":{
        "u":[
          {
            "n":1,
            "v":"B1"
          }
        ]
      }
    },
    {
      "t":7,
      "f0":[
        {
          "g":16,
          "t":0
        }
      ],
      "f1":[
        {
          "g":16,
          "t":1
        }
      ]
    },
    {
      "t":7,
      "f0":[
        {
          "g":5,
          "t":0
        }
      ],
      "f1":[
        {
          "g":5,
          "t":1
        }
      ]
    }
  ]
}`

Let me know if you need any more logs to understand whats going on.

RavenSystem, 2021-04-18:
It is not a bug. RAM is limited. Using a delay between accessory creations will help because some task will be done, freeing memory. That is reason because `"cd"` key is. fatal exception is caused by not enough free RAM (HEAP).

ChristianH99, 2021-04-18:
I understand, thanks.
Could this cause the failure of pairing as well? My phone tells me that the Configuration code is wrong. This is what's in the logs:
Apr 18 19:12:58 [4] New 192.168.44.167:13513 (1/8)
Apr 18 19:12:58 [4] Pairing 1/3
Apr 18 19:13:07 >>> mDNS_recv: wrong size 793 from 192.168.44.248
Apr 18 19:13:12 [4] Pair done 1/3
Apr 18 19:13:12 [4] Pairing 2/3
Apr 18 19:13:14 >>> mDNS_recv: wrong size 1360 from 192.168.44.171
Apr 18 19:13:14 >>> mDNS_recv: wrong size 960 from 192.168.44.171
Apr 18 19:13:16 ! temperature_task: HK pairing
Apr 18 19:13:17 >>> mDNS_recv: wrong size 1360 from 192.168.44.171
Apr 18 19:13:17 >>> mDNS_recv: wrong size 960 from 192.168.44.171
Apr 18 19:13:18 >>> mDNS_recv: wrong size 1362 from 192.168.44.168
Apr 18 19:13:18 >>> mDNS_recv: wrong size 751 from 192.168.44.168
Apr 18 19:13:20 ! temperature_task: HK pairing
Apr 18 19:13:20 >>> mDNS_recv: wrong size 1337 from 192.168.44.172
Apr 18 19:13:20 >>> mDNS_recv: wrong size 734 from 192.168.44.172
Apr 18 19:13:21 >>> mDNS_recv: wrong size 623 from 192.168.44.171
Apr 18 19:13:21 >>> mDNS_recv: wrong size 1412 from 192.168.44.171
Apr 18 19:13:22 >>> mDNS_recv: wrong size 793 from 192.168.44.248
Apr 18 19:13:22 ! [4] Compute pairing code (-2)
Apr 18 19:13:22 [4] Pairing 1/3
Apr 18 19:13:24 >>> mDNS_recv: wrong size 1288 from 192.168.44.171
Apr 18 19:13:24 >>> mDNS_recv: wrong size 847 from 192.168.44.171
Apr 18 19:13:26 [4] Pair done 1/3
Apr 18 19:13:26 >>> mDNS_recv: wrong size 731 from 192.168.44.172
Apr 18 19:13:26 [4] Closing
Apr 18 19:13:26 [4] Closed (0/8)

RavenSystem, 2021-04-18:
`Compute pairing code (-2) ` means no enough memory.

Pairing process needs about 26000 bytes of free HEAP. You can see that value after starting HomeKit Server.

ChristianH99, 2021-04-18:
Puuh, thank you. Maybe I have to split up the devices... I got them paired by removing all the actions from the accessories and putting them back in after pairing. But the system doesn't seem to be very stable like that. Let's see what happens next.

closed 2021-04-18 by ChristianH99