RavenSystem/esp-homekit-devices #1409: Power Strip

okanduzyel, 2021-04-20:
Hi @RavenSystem, 

Will you support power strip accessory?

okanduzyel, 2021-04-22:
I want to create exactly that accessory, not independent outlets. 

[image]
[image]

okanduzyel, 2021-04-23:
the last info:

HAA hasn't support that accessory, yet. 

**Questions and answers:** 
+ Will it is supported?
- Eh, Yes.
+ In the near future? 
- I don't know.

closed 2021-04-23 by okanduzyel

mariusvasss, 2021-04-23:
You know you can group them and use afterwards just like that. Long press on the group and use individual accessories....

okanduzyel, 2021-04-23:
They are pretty close but not the same. I think the definition of a power strip is very simple in the homekit coding structure. Although looks very easy to implement in code, It must be a new accessory. It means, new JSON structure, new definitions, and features, etc.

RavenSystem, 2021-04-23:
Power strip accessory doesn't exist. It is an accessory with 3 or more outlet services.

okanduzyel, 2021-04-23:
> Power strip accessory doesn't exist. It is an accessory with 3 or more outlet services.

I say the same thing. If the definition of outlets individual for HAA like this:

> "a": [{
>     "t": 2,
>     "s": 5,
>     "0": { "r": [{ "g": 12, "v": 0 }] },
>     "1": { "r": [{ "g": 12, "v": 1 }] },
>     "b": []
>   },{
>     "t": 2,
>     "s": 5,
>     "0": { "r": [{ "g": 4, "v": 0 }] },
>     "1": { "r": [{ "g": 4, "v": 1 }] },
>     "b": []
>   }{
>     "t": 2,
>     "s": 5,
>     "0": { "r": [{ "g": 5, "v": 0 }] },
>     "1": { "r": [{ "g": 5, "v": 1 }] },
>     "b": []
>   }]

Homekit can recognize it as a hub. So, never show it power strip. Its code definition must be like this, under one accessory id definition, three or more outlets.

> homekit_accessory_t *accessories[] = {
>     HOMEKIT_ACCESSORY(.id=1, .category=homekit_accessory_category_switch, .services=(homekit_service_t*[]){
>         HOMEKIT_SERVICE(ACCESSORY_INFORMATION, .characteristics=(homekit_characteristic_t*[]){
>             &name,
>             &manufacturer,
>             &serial,
>             &model,
>             &revision,
>             HOMEKIT_CHARACTERISTIC(IDENTIFY, identify),
>             NULL
>         }),
>         HOMEKIT_SERVICE(SWITCH, .primary=true, .characteristics=(homekit_characteristic_t*[]){
>             HOMEKIT_CHARACTERISTIC(NAME, "Socket 1"),
>             &socket_one,
>             NULL
>         }),
>         HOMEKIT_SERVICE(SWITCH, .primary=false, .characteristics=(homekit_characteristic_t*[]){
>             HOMEKIT_CHARACTERISTIC(NAME, "Socket 2"),
>             &socket_two,
>             NULL
>         }),
>         HOMEKIT_SERVICE(SWITCH, .primary=false, .characteristics=(homekit_characteristic_t*[]){
>             HOMEKIT_CHARACTERISTIC(NAME, "Socket 3"),
>             &socket_three,
>             NULL
>         }),
>         HOMEKIT_SERVICE(SWITCH, .primary=false, .characteristics=(homekit_characteristic_t*[]){
>             HOMEKIT_CHARACTERISTIC(NAME, "Socket USB"),
>             &socket_usb,
>             &ota_trigger,
>             &wifi_reset,
>             &task_stats,
>             NULL
>         }),
>         NULL
>     }),
>     NULL
> };

(This code is an example. Not represent HAA)