RavenSystem/esp-homekit-devices #1413: http post request deleting space

NikDevx, 2021-04-28:
`{"c":{"l":13,"n":"Sonoff Switch","b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12}]},"1":{"h":[{"h":"192.168.0.20","p":8123,"m":2,"u":"api/services/script/turn_on","e":"Content-type: application/json\r\nAuthorization: Bearer API_KEY","c":"{"entity_id": "25"}","w":2}],"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0}]}`

Headers should be `Authorization: Bearer  API_KEY`
But when it sending request it's deleting space `Authorization: BearerAPI_KEY`

How I can fix it?

RavenSystem, 2021-08-22:
I can not reproduce it. Could you post API_KEY with all characters replaced by X but two first?

NikDevx, 2021-08-30:
> I can not reproduce it. Could you post API_KEY with all characters replaced by X but two first?

Bearer XxJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiI5MGQ1ZTZiNThmZWQ0MGRkOTBmOWFkYTFjYmIzMzBiOCIsImlhdCI6MTYxNzU5MzkwMywiZXhwIjoxOTMyOTUzOTAzfQ.AJGBMTa8EEW6UlWwfT4mi_Ga9EcuYN62Ld45lx3_0GM

RavenSystem, 2021-09-15:
Are you adding `\r\n` after API_KEY?

NikDevx, 2021-09-15:
It should be?

"e":"Content-type: application/json\r\nAuthorization: Bearer XxJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiI5MGQ1ZTZiNThmZWQ0MGRkOTBmOWFkYTFjYmIzMzBiOCIsImlhdCI6MTYxNzU5MzkwMywiZXhwIjoxOTMyOTUzOTAzfQ.AJGBMTa8EEW6UlWwfT4mi_Ga9EcuYN62Ld45lx3_0GM\r\n"

> Are you adding `\r\n` after API_KEY? It is necessary.

RavenSystem, 2021-09-15:
> It should be?
>
> "e":"Content-type: application/json\r\nAuthorization: Bearer [...]

Yes.