RavenSystem/esp-homekit-devices #1414: Sonoff 4CH R1

lfiocco, 2021-04-29:
Flashed to newest version. However I am not able to turn on more than one accessory on the 4CH. using HomeKit or the buttons. On/off works one at a time. 2 on, does not work.

{"c":{"l":13,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":14}],"s":0},{"0":{"r":[{"g":12}]},"1":{"r":[{"g":12,"v":1}]},"b":[{"g":0}],"s":0},{"0":{"r":[{"g":5}]},"1":{"r":[{"g":5,"v":1}]},"b":[{"g":9}],"s":0},{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":10}],"s":0}]}

any ideas?

lfiocco, 2021-04-29:
Sorry my bad. Discovered what switch S6 does now...

Working fine..please close

closed 2021-04-29 by lfiocco

xtremedays, 2022-10-07:
Hi, I know this was a while ago, but could you explain the 'switch S6'? I'm having the same issue