RavenSystem/esp-homekit-devices #1416: garage door device type on HAA won't work as described

Gandalf8266, 2021-05-02:


kfalkiewicz, 2021-05-12:
Same issue here. Mine stopped reacting on external sensor inputs. Got this code (I don't use magnetic sensor, I take status from gate itself, but it pings when working so that's why I have "f"):
```
{
  "c": {
    "0": {
      "r": [
        {
          "g": 12
        }
      ]
    },
    "l": 13,
    "b": [
      {
        "g": 0,
        "t": 2
      }
    ]
  },
  "a": [
    {
      "0": {
        "r": [
          {
            "g": 12,
            "v": 1,
            "i": 0.5
          }
        ]
      },
      "1": {
        "a": 0
      },
      "2": {
        "r": [
          {
            "g": 12,
            "i": 0.5,
            "v": 1
          }
        ]
      },
      "3": {
        "a": 2
      },
      "t": 40,
      "d": 1,
      "f2": [
        {
          "g": 5,
          "t": 0,
          "f": 120
        }
      ],
      "f3": [
        {
          "g": 5,
          "t": 1,
          "f": 120
        }
      ],
      "s": 1,
      "b": []
    }
  ]
}
```

claire078, 2021-06-06:
> I tried to configure a Sonoff Basic as a garage door opener. I used newest firmware 5.0.18. I connect that [magnet contact ](https://de.aliexpress.com/item/4000209658438.html?spm=a2g0o.productlist.0.0.110ca6a6EiEdt8&algo_pvid=563f6b6c-9de9-46e2-ba09-7913ce058701&algo_expid=563f6b6c-9de9-46e2-ba09-7913ce058701-12&btsid=2100bdec16197631377953118e2934&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)(reed contact) to GND and GPIO14 (the last pin right to GND) and I used the default configuration:
> {
> "c":{"l":13,"b":[{"g":0,"t":5}],"0":{"r":[{"g":12}]}},
> "a":[{
> "t":40,
> "d":18,
> "0":{"r":[{"g":12,"v":1,"i":0.5}]},
> "1":{"a":0},
> "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":1,"i":0.5},{"g":12,"v":1,"i":2}]},
> "3":{"a":2},
> "f3":[{"g":14,"t":0}],
> "f4":[{"g":14,"t":1}]
> }]
> }
> The garage door appears in HomeKit but the Hombrige Icon are always aktiv an shows "closing":
> [image]
> 
> I don't have any Ideas to fix that Problem.
> The Reed contact works, when I'am approaching the magnet, the contact closes, (testet with a multimeter).

"f3":[{"g":14,"t":0}],
"f4":[{"g":14,"t":1}]

Same input for two states.

closed 2021-07-07 by stale[bot]