RavenSystem/esp-homekit-devices #1418: How to get the IR code to control a device

WindowsMEMZ, 2021-05-04:
The following is from Baidu translation, translated from Chinese to English.
I want to control my thermostat through this project, and I'm only one step short of getting the IR code sent by the remote control to the thermostat, but I can't get it. Is there any way to receive and decode the IR code from the remote control?

RavenSystem, 2021-05-06:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#ir-capture-gpio
https://github.com/RavenSystem/esp-homekit-devices/wiki/HAA-Infra-Red#how-haa-supports-infra-red-control

WindowsMEMZ, 2021-05-08:
I see. thank you. But now my esp8266 can't enter setting mode. I've tried the operation switch eight times, and I don't fill in JSON, but none of them can let me into setup mode, just like the video I gave it. [](https://share.icloud.com/photos/0g5_GOZuHcYC-rfRByxs2sr2g)

The device I use is the nodemcu development board of esp8266, whose chip is esp-12F. I inserted an infrared receiving diode into GPIO5 and an infrared emitting diode into GPIO2.@RavenSystem

WindowsMEMZ, 2021-05-09:
I see. thank you. But I don't know how to monitor the serial port. I have a USB to TTL converter. I connected TXD on ESP to RXD on the converter, RXD to TXD, and connected GND. But I don't have a serial port monitor on Windows. is there any software to monitor the serial port? If not, I also have a Linux-based Windows subsystem, on which can I monitor serial ports? @RavenSystem

closed 2021-05-22 by WindowsMEMZ