RavenSystem/esp-homekit-devices #1419: IR CODE LG AIR Conditioner

ur5zeb, 2021-05-05:
Hello
Help me please with IR codes for LG air conditioner

ur5zeb, 2021-05-09:
I scan the IR remote
OFF
Packets: 59
Standard Format
+ 8669 - 4107 +  486 - 1569 +  431 -  594 +  432 -  585 +  431 -  586 +  431 - 1610 +  433 -  584 +  433 -  587
+  431 -  593 +  431 - 1605 +  432 - 1608 +  430 -  594 +  433 -  583 +  431 -  589 +  431 -  588 +  431 -  592
+  431 -  586 +  433 -  587 +  431 -  588 +  432 -  589 +  433 -  586 +  431 -  589 +  431 - 1612 +  459 -  557
+  457 - 1584 +  431 -  592 +  431 -  584 +  431 -  589 +  433 - 1610 +  431
HAA RAW Format
T?I*ANC!ACAiACAhACAhACC?ACAgACAhACAiACC=ACC=ACAiACAgACAhACAhACAiACAhACAhACAhACAhACAhACAhACC?AHAbAHC$ACAiACAgACAhACC?AC

ON Cooling
Packets: 59
Standard Format
+ 8658 - 4150 +  493 - 1558 +  432 -  591 +  465 -  552 +  433 -  589 +  431 - 1608 +  433 -  586 +  432 -  588
+  431 -  589 +  429 -  593 +  465 -  551 +  432 -  588 +  432 -  588 +  432 -  591 +  465 -  551 +  432 -  589
+  408 -  611 +  432 -  591 +  465 -  552 +  431 - 1609 +  431 - 1609 +  432 -  588 +  430 - 1612 +  463 -  554
+  433 -  587 +  431 -  588 +  431 - 1607 +  433 - 1608 +  431 - 1609 +  432
HAA RAW Format
T)J0AOC+ACAiAJAaACAhACC=ACAhACAhACAhABAiAJAaACAhACAhACAiAJAaACAh0<AmACAiAJAaACC=ACC=ACAhACC?AIAaACAhACAhACC=ACC=ACC=AC

I read the protocol and IR command
But they don't work. What am I doing wrong?
Protocol: T?I*ANC!ACAiAC
command: aCaCbKaAaCa

ur5zeb, 2021-05-09:
Correct protocol  
"p":"T?I*AGC?AJAiAC",
"0":{"i":[{"c":"aCaCbKaAaCa"}]},     //OFF
"3":{"i":[{"c":"aCaHaBaAaBaDaA"}]},  //Heating 25 С FAN MAX
"4":{"i":[{"c":"aCaMbAaCc"}]},           //Cooling 18 С FAN MAX

closed 2021-06-09 by stale[bot]

sercitecotf, 2023-03-28:
have you been able to configure the lg air?