RavenSystem/esp-homekit-devices #1422: Window blinds

ooomakon, 2021-05-08:
Good afternoon, help me compose json for blinds with a 28BYJ-48 motor and a ULN2003 driver, there is an understanding with a conventional motor, but I can’t figure it out.

joddye2, 2021-05-15:
I think that the window covering example only for motorized shades witch have built in max, min position sensor. 
I saw an other HomeKit firmware but its for 12V type of 28byj and another driver: https://www.studiopieters.nl/esp8266-homekit-blinds-2/
..But If u make a json for 28byj please share with me :)

closed 2021-06-15 by stale[bot]

exrector, 2021-10-01:
> Добрый день, помогите составить json для штор с мотором 28BYJ-48 и драйвером ULN2003, с обычным мотором понимание есть, но разобраться не могу.

Did you manage to solve something in this matter?