RavenSystem/esp-homekit-devices #1423: Sonoff Basic RFR3 Stuck Auto-updating

DarrenBishop, 2021-05-09:
Hi all, need some help with my Basic RFR3.

I recently flashed Tasmota then managed to get HAA flashed, all OTA, using [itead/Sonoff_Devices_DIY_Tools](https://github.com/itead/Sonoff_Devices_DIY_Tools) and then using firmware upgrade [`homekit-from-tasmota.bin/zip`](https://github.com/RavenSystem/esp-homekit-devices/files/4427799/homekit-from-tasmota.zip) from this [thread](https://github.com/RavenSystem/esp-homekit-devices/issues/683).

I did this at my parents' on my dad's Win machine using the Sonoff_Devices_DIY_Tools `tool/tool_01DIY85(3.3.0).exe` as I could not get the `code/tool_01DIY85.py` to work on Mac Big Sur M1 with Rossetta2 and shaded Homebrew, Python 3.9 and PIP  (i.e. `arch -x86_64 /bin/bash`) ... I really tried.

Note: I am a software (data) guy... not so much (at all) an electronics/firmware guy.

Now when I plug my Basic RFR3 in, I get the following logs:

```bash
└▪ nc -kulnw0 45678
Select ROM 1
Restarting...
HAA OTA Version: 2.2.2
INIT
DNS OK
Current ROM is 0
Set version: 0.0.0
HAA firmware
Repo='RavenSystem/haa' Version='0.0.0' File='main.bin'
Current ROM is 0
Get version RavenSystem/haa
NEW CONNECTION LocalPort=ecd4 DNS IP: 140.82.121.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK

GET /RavenSystem/haa/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for Current ROM is 0
Compare
5.0.18 with 2.2.2 = 1
RavenSystem/haa@version:"5.0.18"

STARTING

Get version RavenSystem/haa_ota
NEW CONNECTION LocalPort=ecd5 DNS IP: 140.82.121.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK

GET /RavenSystem/haa_ota/releases/latest HTTP/1.1
Host: github.com

HTTP returns 302 for Current ROM is 0
Compare
4.5.0 with 2.2.2 = 1
RavenSystem/haa_ota@version:"4.5.0"
Current ROM is 0
Running BOOT
Get hash
DOWNLOADING
GET /RavenSystem/haa_ota/releases/download/4.5.0/main.bin.sig HTTP/1.1
Host: github.com

NEW CONNECTION LocalPort=ecd6 DNS IP: 140.82.121.4 local..OK remote..OK SSL..OK set_fd to github.com port 443..OK
sent OK
HTTP returns 302
NEW CONNECTION LocalPort=ecd7 DNS IP: 185.199.108.154 local..OK remote..OK SSL..OK set_fd to github-releases.githubusercontent.com port 443..OK
Select ROM 1
Restarting...
```

From here it just hangs...all night long.

Note that I believe the firmware is trying to download
> https://github.com/RavenSystem/haa_ota/releases/download/4.5.0/main.bin.sig

...which is redirected; downloading from Chrome redirected to

> https://github-releases.githubusercontent.com/334639627/7154f500-66e1-11eb-9c94-ce5f6f75c92f?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20210509%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20210509T090433Z&X-Amz-Expires=300&X-Amz-Signature=363b0d9277596beb0b71dd23d8383dbeb29bbd81c23551b6ee633e8311cee4cd&X-Amz-SignedHeaders=host&actor_id=47794&key_id=0&repo_id=334639627&response-content-disposition=attachment%3B%20filename%3Dmain.bin.sig&response-content-type=application%2Foctet-stream

...is this URL (at 518 characters) even able to be handled by HAA?

I tried switching off my WiFi router and power-cycling the Basic RFR3, but never get the `HAA-XXXXXX` SSID back.

Any suggestions for resolving this?

Feel this is resolvable so have already ordered an extra bunch of Basic R3's  and Mini R2's

pilot1981, 2021-05-09:
I wired it to flash. Sonoff R3RF works fine with HAA!

DarrenBishop, 2021-05-09:
I have ordered some electronics to flash OTW (i.e. over-the-wire) but was hoping to avoid all that.

Would still be nice to understand why this OTA process seems to be failing?

RavenSystem, 2021-06-04:
You are using a very old HAA Installer file. You must use last fullhaaboot.bin from this current repo to ensure it will work.

closed 2021-06-04 by RavenSystem

DarrenBishop, 2021-06-04:
I've used fullhaaboot.bin on subsequent devices successfully.

I now have both a WiFi and a Zigbee Sonoff devices connected to Home Assistant, along with a Hive motion sensor; also all visible in Apple Home.

That device mentioned earlier is still zombied though; don't now how to recover it - seems happy in its own little world, just can't communicate with it. Having opened it up, there doesn't seem to be obvious/easy serial flash connections.

Even with old firmware, surely there's an OTA way to recover it i.e. to upgrade the firmware?

DarrenBishop, 2021-06-04:
Also, if anyone can explain the netcat log?
It's not obvious what is going wrong: is it running off ROM 0, downloading new firmware to ROM 1, trying to restart booting from ROM 1 but failing?

DarrenBishop, 2021-06-17:
So I managed to re-flash all zombied devices OTW i.e. with a USB-to-FTDI device. I got a 3D print of the [rig off thingiverse](https://www.thingiverse.com/thing:3767667) to reclaim the minis and got inventive with some [Dupont jumper wires](https://www.amazon.co.uk/gp/product/B01EV70C78/ref=ppx_yo_dt_b_asin_image_o01_s00?ie=UTF8&psc=1).

On with the installation.