RavenSystem/esp-homekit-devices #1425: Sprinkler issue sonoff 4ch r3 pro

franklindee, 2021-05-13:
Hi i've set up a sonoff 4ch pro R3 with this json

{"c":{"o":0,"m":60,"l":13,"i":0,"f":10,"b":[{"g":0,"t":5}]},"a":[{"0":{"r":[{"g":12,"v":0}]},"1":{"r":[{"g":12,"v":1}]},"t":20,"b":[{"g":0,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":5,"v":0}]},"1":{"r":[{"g":5,"v":1}]},"t":20,"b":[{"g":9,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":4,"v":0}]},"1":{"r":[{"g":4,"v":1}]},"t":20,"b":[{"g":10,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]},{"0":{"r":[{"g":15,"v":0}]},"1":{"r":[{"g":15,"v":1}]},"t":20,"b":[{"g":14,"t":1,"p":1}],"s":0,"w":1,"d":0,"f0":[]}]}

if i try to set up an automation he sprinkler doesn't show and if i open the single item i cannot plan any activation

do you know how to solve it?

mariusvasss, 2021-05-13:
Use Eve and make an individual scene for every sprinkler. Then use te automation, you'll find the Eve scenes to use them....

franklindee, 2021-05-13:
i've tryed to use the eve app but when i open the home app the automation is empty will this work anyway?
thank you

mariusvasss, 2021-05-13:
You will create a scene to turn on the sprinkler￼ 1 with the desired time. Name it scene 1. After that you will create another scene just like the first one but with second sprinkler and so on for all 4. ￼
After that use home to create automation based on time, use convert to shortcut and select scene 1, use wait in shortcut Equal with the time of the sprinkler (5 mins or so) then scene 2, again wait equal with sprinkler time and so on for all 4

franklindee, 2021-05-13:
i will try thanks for the explanation, i though there was a way that Jose could modify the sprinkler accessory to be able to set on the single item :)

mariusvasss, 2021-05-13:
I found this for me to be the best solutions. They work in row one after each other...

closed 2021-05-17 by RavenSystem