RavenSystem/esp-homekit-devices #1430: shelly 2.5 potentially overheating

RAZRu, 2021-05-21:
Is the problem described here:

https://savjee.be/2020/11/shelly25-esphome-potential-fire-hazard-fix/#:~:text=If%20you%20have%20flashed%20your,creating%20a%20potential%20fire%20hazard!&text=It's%20a%20bad%20ESPHome%20config%20causing%20a%20short%20circuit%2C%20yikes!

fixed in template from Device Database?
Because I not see any g;16 in json

Switch with external switches, power monitoring, and internal temperature monitoring from HAA Manager:
`{"c":{"l":0,"b":[{"g":2,"t":5}],"ic":[[14,12,100]]},"a":[{"0":{"r":[{"g":4}]},"1":{"r":[{"g":4,"v":1}]},"b":[{"g":13,"p":0},{"g":13,"t":0}],"es":[{"0":{"r":[{"g":15}]},"1":{"r":[{"g":15,"v":1}]},"b":[{"g":5,"p":0},{"g":5,"t":0}]},{"t":75,"cd":1.5,"n":2,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":0.006097560976},{"t":75,"n":3,"dt":[0,56],"vf":0.0000382602,"vo":-0.068,"cf":0.00000949523,"co":-0.017,"pf":-0.006097560976},{"t":22,"h":2,"n":5,"j":10,"y0":[{"v":75,"r":1,"0":{"m":[[1],[2]]}}]}]}]}`

Thanks in advance

RavenSystem, 2021-05-21:
There is not problem with Shelly 2.5 overheat. All undeclared GPIOs are initialized automatically as INPUT to avoid hardware damage.

pilot1981, 2021-05-21:
@RavenSystem automation to shutdown Shelly in case of overheat?

RAZRu, 2021-05-21:
> @RavenSystem automation to shutdown Shelly not needed in case of overheat?

I think this is not redundant, especially with a high-current load

RavenSystem, 2021-05-21:
In JSON, you can see that device will turn off both relays is temperature is 75C or more.

RavenSystem, 2021-05-21:
> > @RavenSystem automation to shutdown Shelly not needed in case of overheat?
> 
> I think this is not redundant, especially with a high-current load

You are right, it is not redundant.

closed 2021-05-22 by RavenSystem