RavenSystem/esp-homekit-devices #1432: Shelly RGBW2 - always switch on to white colour

kompak23, 2021-05-24:
Hello, 
I have an issue, I use Shelly RGBW2 with RGB LED strip as main lighting. 
I use physical switch on the wall to tur on the light, but I need to turn on always white colour.
Is it possible? I have found that there is "it" parremater which can set default colour, but it works only after booting device. 
When someone change colour to other and then turn off light, and after I press switch, the colour is remebered. 

Can somebody help me? 

Thanks, 
kompak23

mariusvasss, 2021-05-24:
Use a separate accessory only for white with the switch, and colours only in the home app... hope this helps.

RavenSystem, 2021-06-04:
@kompak23 enter Discord chat and ask there. Other user did that. It is possible using service notifications.

closed 2021-06-04 by RavenSystem

kompak23, 2021-06-15:
> Use a separate accessory only for white with the switch, and colours only in the home app... hope this helps.

how to set up accessory as white bulb, when there is RGB light strip?