RavenSystem/esp-homekit-devices #1433: Sonoff MiniR2

Murdrock, 2021-05-26:
Has anybody managed to hard flash a sonoff Minir2? I have the button pressed while inserting the USB but NodeMCU doesn't seem to do anything. I have flashed plenty V1 Mini. Is there a different procedure to make it go to Flash mode?

Thanks in advance

iondarie, 2021-05-26:
I have on MacOS using NodeMCU PyFlasher, held the button pressed while inserting the usb and after that it found the device and started flashing.

Murdrock, 2021-05-28:
> I have on MacOS using NodeMCU PyFlasher, held the button pressed while inserting the usb and after that it found the device and started flashing.

Next day I tried it, it workd. I donw know what was wrong in the first place. So far it seems that the procedure is the same with V1. 

Thanks for all responces!

closed 2021-06-04 by RavenSystem