RavenSystem/esp-homekit-devices #1434: Flashing Sonoff mini R2 OTA with HAA

jcastro-design, 2021-05-26:
Hello!

I'm new to home automation, but decided to give it a go and acquired some Sonoff Mini R2 to use with HAA firmware for HomeKit compatibility. (My only previous experience was with Sonoff basic devices flashed via ftdi with custom firmware)

I've tried to flash the Mini by putting it in DIY mode and then using the Sonoff DIY tool (https://github.com/itead/Sonoff_Devices_DIY_Tools) to flash with fullhaaboot.bin, which resulted in bricking the device.. (firmware 3.6.0 on the mini)

From what I understood afterwards, the tool is not compatible anymore with 3.6.0 firmware, so my question is:
Is there another tool for flashing OTA the Sonoff mini R2 with HAA? Where can I read about this process step-by-step, starting with the factory firmware?

Thanks in advance for any help!

Also, if someone has an updated and comprehensive (noob edition) link to a guide I would be very grateful. (most of the videos / tutorials I keep finding about flashing sonoff mini r2 are outdated)

belkone, 2021-05-26:
I don't know why, but it is not possible to flash Sonoff OTA straight to HAA.
You can flash to Tasmota first and then flash to HAA from the Tasmota web panel.
Reference: https://tasmota.github.io/docs/Sonoff-DIY/

ex725, 2021-05-26:
How to go from haa to tasmota.

jcastro-design, 2021-05-27:
Thank you @belkone and @Daveblanche for your answers and help.

I’ve managed to make everything work out ok following parts of a guide I’ve found here (https://blog.lukaskukacka.com/smarthome/2020/09/09/sonoff-mini-firmware-3.6-flashing-tasmota.html), and a few bits from the tasmota guide you’ve posted above.

For future documentation, and anyone else who’s attempting the same, here’s how I’ve managed to get everything up and running:

**Step one: Place your Sonoff Mini r2 in DIY mode**

If you’re using a Mini R2 (internal antenna) simply connect it to power using a simple AC plug and wire, and afterwards long-press the button on the device until the LED flashes continuously.

From mobile phone or PC WiFi settings, an Access Point of the device named ITEAD-XXXXXXXX will be found, connect it with default password 12345678

After a successful connection, open a browser window and access http://10.10.7.1/

Fill in WiFi SSID and password. Once successfully connected, the device is in DIY mode and connected to your WiFi network. 

Check your network devices to find out its assigned IP address and take note of it (you will need this on the next step)

**Step two: Validate if OTA is unlocked on your mini**
Using Postman (or other software you prefer), you must make a POST request to your device.

url: http://<yourDeviceIPAddress>:8081/zeroconf/info
Body:
{
  "deviceid": "",
  "data": { }
}

(If in return you see otaUnlock is true you can proceed to step five)

{
...
  "otaUnlock": true,
  "fwVersion": "3.6.0",
...
}

**Step three: Unlock OTA flash on your sonoff mini**
Using Postman make a POST request to your device:

url: http://<yourDeviceIPAddress>:8081/zeroconf/ota_unlock
Body:

{
  "deviceid": "",
  "data": { }
}

**Step four: re-check if OTA unlock is enabled on your sonoff mini**
Before moving forward, it’s always best to be sure everything worked as intended.
Using Postman make a POST request to your device:

url: http://<yourDeviceIPAddress>:8081/zeroconf/info
Body:
{
  "deviceid": "",
  "data": { }
}

**Step five: install tasmota lite on your device**
If everything went as expected, its now time to flash tasmota-lite into the device using Postman. 
Make a POST request to your device:
 url: http://<yourDeviceIPAddress>:8081/zeroconf/ota_flash

Body: 
{
  "deviceid": "",
  "data": {
    "downloadUrl": “YOUR TASMOTA LITE BIN FILE URL goes here”,
    "sha256sum": “SHA256 FOR YOUR FILE HERE”
  }
}

You can get tasmota links from this setup server: http://sonoff-ota.aelius.com

**Step six: setup tasmota**
After flashing the device with tasmota-lite firmware it will create a new Wireless AP named something like tasmota_XXXXXX. Connect to it, and you’ll be redirected to the config page (if it doesn’t open a browser window with the configuration page you can access it on http://192.168.4.1 )

Select your network from the list and insert your credentials on the configuration page, click Save and the device will restart to apply.

The mini is now connected to your WiFi and flashed with tasmota-lite. Scan your network for its IP address and take note of it. You will need it for the next step.

**Step seven: flash HAA on your device using tasmota web UI**
Download fullhaaboot.bin from the guide on this page: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation

Enter your device’s IP address (from the previous step) on a new browser window to access the tasmota-lite config page. 

From here, we’ll use the Firmware Upgrade function to flash HAA into our tasmota-lite device, by selecting the option Ugprade by file upload and selecting fullhaaboot.bin as our file.

The flashing process will take a few minutes, and you must wait until everything completes. When everything is complete you’ll see a new AP on your WiFi networks named HAA- followed of last MAC address.

Join the new AP, open a browser window and type http://192.168.4.1:4567 to access HAA configuration page.

**Step eight: configure HAA on your sonoff mini and add the proper json config to it**
From here onwards you can follow the configuration instructions on the HAA Wiki page, from point 5.
https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation

On the HAA configuration page, similarly to what you’ve done with tasmota-lite, select your wifi network and insert your credentials.

Additionally, you need to add a configuration JSON to your accessory, so that it can be added to Home App later. More information on the JSON config is available here: https://github.com/RavenSystem/esp-homekit-devices/wiki/Configuration

Personally, I have found easier to use this tool to generate the JSON config: https://glumb.github.io/haa-configurator/

You can later access Setup mode, where you will be able to change configuration or even update firmware OTA.

Hit Save and now you must wait around 4 minutes for the device to restart and all configurations to apply.
Don’t rush, let it do its thing.

**Step nine: add the new accessory to HomeKit**
If everything went as expected you are now able to add your new accessory to the Home App, by scanning the QR code available here: https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#homekit-setup

And that’s it! Hopefully this guide will come in handy for anyone in the same situation as I was. Thanks again to @belkone  and @Daveblanche for your help, and credits go to all the authors of the above mentioned documentation.

Daveblanche, 2021-05-27:
Nice work!!

RavenSystem, 2021-05-27:
@jcastro-design I added your comment to Wiki Installation process:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Installation#third-party-installation-tools

Many thanks!

closed 2021-05-27 by RavenSystem

fbe28, 2022-03-14:
I have followed these steps with a Sonoff Mini R2 and it worked up until the point of adding the device to Homekit. Unfortunately, the device disconnected and I cannot seem to get into setup mode with it reliably at all. I have tried toggling the switch 8 times and that works very sporadically..

The times I did managed to get into the settings, unfortunately the page/network would become unavailable quickly and I couldn't get it to work :/

I even bought the iOS app which, at first run, made things go back to working again, but it suddenly stopped again.

Has anyone run into this before? I really love the work that has been put into this and was so thrilled when it started working.. I just want to enjoy using this super cool tool!

Thanks in advance,

Fernando

fbe28, 2022-03-14:
> I have followed these steps with a Sonoff Mini R2 and it worked up until the point of adding the device to Homekit. Unfortunately, the device disconnected and I cannot seem to get into setup mode [...]

UPDATE:
After trying many times, I managed to access the Setup mode and very quickly reset the settings before the connection dropped.

I made all the changes and added the JSON config, connected to the network, but unfortunately the device now does not get recognized inside Homekit or in the HAA Manager.

ggpedraza, 2022-06-24:
Thank you! I used this tutorial some time ago with success, however I can't remember what I did, but I wasn't exactly the same. I think I used DIY tool to update the firmware. Now I'm trying to do it again, but I stuck sending request to flash the firmware (step five). After waiting some minutes, Postman says "Error: socket hang up". Any ideas?

Sorry for reopen the thread, but I don't know how to solved it. Any help would be appreciated!

MatSkovbo, 2023-01-23:
This has been working flawlessly for a long time. I recently got a new wifi-router, with a new SSID and password. Is there an easy way to switch it over to a new network, or should the whole process be repeated?

joel-daros, 2023-07-02:
I tried the suggested steps by @jcastro-design but after I flashed the HAA firmware through Tasmota update firmware page, I bricked my Sonoff Mini R2. I flashed the ESP8266 [tasmota2haa.bin.gz](https://github.com/RavenSystem/haa/releases/latest/download/tasmota2haa.bin.gz) suggested in the README.
I waited about 15 minutes and the HAA network never shows. No light and no network. 

What I did wrong, and most important, Is there anything that I can do to back it alive again?