RavenSystem/esp-homekit-devices #1435: How to go from haa to tasmota.

ex725, 2021-05-26:
all my products are banned.

This product gives an error saying inappropriate. Homekit id reset does not work.

How to go from haa to tasmota.

peros550, 2021-05-27:
You would need to flash by wire

ex725, 2021-05-27:
Custom server for updates:
Can't we do it by uploading from this section.

RavenSystem, 2021-05-27:
Only with wires.

closed 2021-05-27 by RavenSystem

daryelv, 2025-01-22:
Given that the developer has no interest in you leaving his ecosystem, he's not going to provide a way to do it.  Unfortunately this means you either 

1) buy all new devices - this would run me around $400-$500 for the devices plus all the time to replace the firmware on each new device with Tasmota, and finally all the time to physically replace the HAA device with the new Tasmota flashed one.  I'd estimate around 20-32 hours of labor to do this with the amount of devices and the locations they're in.

2) refresh existing devices - since you need to use hardware to reflash, this involves uninstalling the device, re-flashing, re-installing.  This isn't even possible on some devices that don't come apart (small outlets for example).  I'd estimate the hours to be around 48-56 total for all the devices in my setup.

3) a developer here embraces "Open Software" standards or provides a way to flash Tasmota over HAA.  Even if it was an old version of Tasmota that was 'blessed' with the signed keys that are required for flashing, that would at least let us reflash something like a ceiling fan control (I have 2) which are very difficult to get to.  

Considering this disclaimer was not added until July 14, 2022, I feel that a tool to allow flashing HAA back to Tasmota should be made available to anyone that has been stuck using HAA since before July 14, 2022. 

[image]