RavenSystem/esp-homekit-devices #1438: Flash corrupted after power outage

pbendersky, 2021-05-29:
After a brief power outage, my device now reboots in setup mode, with a red warning that reads: `FLASH corrupted! Reset settings recommended`.

The error seems to be [this one](https://github.com/RavenSystem/esp-homekit-devices/blob/2199549a2a69b25a35392a5af7bbe484e751438b/devices/HAA/setup_mode/src/wifi_config.c#L499), but I'm not sure what triggers `context->param` to be `1`, nor what that means.

I've tried (to no avail):
- Reset Settings (nothing happened after reboot)
- Reset HomeKit ID (nothing happened after reboot)
- Changing the WiFi configuration to a secondary SSID I have (the setting was not saved, it rebooted connected to the previous SSID)
- Updating OTA (it did not update).

I'm running v5.0.16, with HAA OTA v4.7.2.

Due to the location of the device, I'm trying to fix it without having to plug a serial adapter... Any idea what can be causing this and if it's fixable remotely?

My JSON config is as follows:
```
{
  "c":{"l":2,"b":[{"g":5,"t":5}],"0":{"r":[{"g":4}]}, "ttl": 60},
  "a":[{
    "t":40,
    "d":18,
    "vs": 1,
    "0":{"r":[{"g":4,"v":1,"i":0.5}]},
    "1":{"a":0},
    "2":{"a":0},
    "3":{"a":0},
    "10":{"a":0}
  },
  {
    "t":1,
    "s": 0,
    "1": {"s": [{ "a": 1}] }
  }]
}
```

pilot1981, 2021-05-29:
Same thing happens to me on 5 Shelly with 5.0.18 at same time.

the only solution was to flash again using wire!

I had some problem to flash it again…you will have to erase better the flash…

In one case I had a Shelly 2.5 broken: one output port have always current (a relay problem)

davebra, 2021-06-01:
Same here, with HAA 6.0.4 on a ESP-01 configured as switch.
Only solution was re-flash.

RavenSystem, 2021-06-04:
If erasing all data doesn't work, only solution is erase and reflash with wires.

closed 2021-06-04 by RavenSystem

RavenSystem, 2022-10-04:
To avoid future issues, be sure that devices have right [SPI flash mode](https://github.com/RavenSystem/esp-homekit-devices/wiki/Setup-Mode#Flash) selected.