RavenSystem/esp-homekit-devices #1441: Can HAA Manager work with a bridged device?

alexcr28, 2021-06-01:
I have a HAA device connected to Home Assistant using the "HomeKit Controller" configuration which is connected to my Apple HomeKit environment using the HomeKit bridge configuration.

HAA Manager can see the device as a bridge accessory of the HASS bridge but I can't control any of the custom settings (update/setup) like I can with HAA devices connected directly to my Apple HomeKit environment.

Is this a limitation of the device being bridged or can HAA Manager be updated to take care of this scenario?

I want to use Home Assistant for automations since I find that automations in HomeKit don't always run.

andrewbpowers, 2021-06-02:
Hello!

I would also like to have a solution for exactly the scenario you describe.

Take care...

RavenSystem, 2021-06-04:
HAA Manager can work with bridge service, but only if bridge exposes all custom HomeKit characteristics needed; Home Assistant doesn't.

closed 2021-06-04 by RavenSystem

andrewbpowers, 2021-06-04:
Thanks for the feedback!

So HAA/HAA Manger is with Home Assistant absolutely not to useable, because you have to disassemble everything if you have a problem or you want change the configuration and must re-flash the hardware.