RavenSystem/esp-homekit-devices #1442: Jason for Sonoff Dual R3 shutter blinds

Travelworm, 2021-06-05:
Can I use the Shelly 2.5 Jason for Sonoff Dual R3, or is there a specific jason for this device to work as shutter blind switch?

mariusvasss, 2021-06-05:
Nope. It will not work. Different GPIO. You have to modify the json to match GPIO for the relay and buttons.

Travelworm, 2021-06-05:
Thanks. What about the Sonoff 2ch jason for blinds with external switch?

I really appreciate your help. 

Erik Richter

mariusvasss, 2021-06-05:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Devices-Database

Take a look here

mariusvasss, 2021-06-05:
You have to soleder some wires on the board for the external buttons.

mariusvasss, 2021-06-05:
*solder

Travelworm, 2021-06-05:
The dual R3 already has S1 and S2 connection for the external switch. 

Erik Richter

mariusvasss, 2021-06-05:
Then use that json for dual r2 and change your gpio to adapt it.

Travelworm, 2021-06-05:
Ok thanks

Erik Richter

mariusvasss, 2021-06-05:
You just have to find the relays and the buttons gpio and change them in that json to work.

mariusvasss, 2021-06-05:
Relay 1 is GPIO 27
Relay 2 is GPIO 14
S1 is GPIO 32
S2 is GPIO 33

mariusvasss, 2021-06-05:
{"c":{"l":13,"b":[{"g":32,"t":5}]},"a":[{"t":45,"o":18,"c":14,"f":70,"0":{"r":[{"g":27,"v":1},{"g":14}]},"1":{"r":[{"g":27},{"g":14,"v":1}]},"2":{"r":[{"g":27},{"g":14}]},"3":{"r":[{"g":27,"i":1},{"g":14}]},"4":{"r":[{"g":27},{"g":14,"i":1}]},"f3":[{"g":32}],"f4":[{"g":33}]}]}

try this. i made it for you

mariusvasss, 2021-06-05:
hope it works

mariusvasss, 2021-06-05:
Don't forget to adjust your opening and closing time 'c' and 'o' in seconds

Travelworm, 2021-06-05:
Thank you

Erik Richter

closed 2021-07-05 by stale[bot]