RavenSystem/esp-homekit-devices #1445: Garage door opens when booting

astoncura, 2021-06-09:
Hi,
I am struggling with my garage door configuration. Everything works just fine apart from booting. Every time I boot the Wemos D1 Pro the garage door opens.

{
  "c":{"l":13,
  "b":[{"g":0,"t":5}],
  "0":{"r":[{"g":12}]}},
  "a":[{
    "t":40,
    "d":18,
    "xa":0,
    "0":{"r":[{"g":12,"v":0,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":0,"i":0.5},{"g":12,"v":0,"i":2}]},
    "3":{"a":2},
    "f3":[{"g":14,"t":0}],
    "f4":[{"g":14,"t":1}]
  }]
}

astoncura, 2021-06-10:
I found out that "0":{"r":[{"g":12}]}} sets the initial state to LOW on boot. However I have to run the relay I am using on LOW at activate. How can I change the initial state to HIGH when booting?

QDUNI, 2021-06-10:
Why are you using "0":{"r":[{"g":12}]}}, before the "a":[{}] ? If you really need it try "0":{"r":[{"g":12, v:1}]}},

astoncura, 2021-06-10:
Well, I actually used it because it's used in the example:
https://github.com/RavenSystem/esp-homekit-devices/wiki/Garage-Door

QDUNI, 2021-06-10:
Try what I just send you or leave it out before the "a":[{}]

astoncura, 2021-06-10:
Thanks for your help. I tried both but the device stays offline for both options.

QDUNI, 2021-06-10:
What device are you using ?

astoncura, 2021-06-10:
I am using a Wemos D1 Mini Pro.

QDUNI, 2021-06-10:
I hope you noticed I made a typo, you have to do this before the "a":[{}]. 

"0":{"r":[{"g":12, "v":1}]}} 

Make sure your json is correct by checking it on a online json validator and covert it to compact. 

https://jsonformatter.curiousconcept.com/#

astoncura, 2021-06-10:
Thanks, its working now. Unfortunately the relay still gets pulled for a few milliseconds, when booting...

astoncura, 2021-06-11:
Thank you for your help. Unfortunately it is still not working. The esp still boots on low. I also tried different gpios with the same result. Here is my current config:
{
  "c":{"l":13,"b":[{"g":0,"t":5}]},"0":{"r":[{"g":12,"v":1}]},
  "a":[{
    "t":40,
    "d":18,
    "0":{"r":[{"g":12,"v":0,"i":0.5}]},
    "1":{"a":0},
    "2":{"r":[{"g":12,"i":1.5},{"g":12,"v":0,"i":0.5},{"g":12,"v":0,"i":2}]},
    "3":{"a":2},
    "f3":[{"g":14,"t":0}],
    "f4":[{"g":14,"t":1}]
  }]
} 

what am I missing?

QDUNI, 2021-06-11:
Did you try gpio 2 this one needs to be high at boot in order to boot ?

closed 2021-07-11 by stale[bot]

astoncura, 2021-07-14:
So I tried gpio 2 and it didn't work. I just bought a new relay that pulls on high and it works fine.

Thanks for your help @QDUNI !!